ANYbotics/grid_map #33: origin conflict when converting from grid map to occupancy grid

bit-pirate, 2016-02-10:
I started playing around with this library and as a first test I implemented a "round trip conversion": `nav_msgs::OccupancyGrid`-> `grid_map_msgs::GridMap` -> `grid_map::GridMap` -> `nav_msgs::OccupancyGrid`.

The first conversion is done by my own implementation (couldn't find an implementation in this library), the second is using `grid_map::GridMapRosConverter::fromMessage` and the last `grid_map::GridMapRosConverter::toOccupancyGrid`. 

The result looks fine, except 1) the costmap is rotated and 2) there is an offset in the origin's position.

Starting with the **first issue**, the original origin's quaternion is:

```
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

The quaternion of the intermediate grid map is the same:

```
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

But the final quaternion is different:

```
    orientation: 
      x: 0.0
      y: 0.0
      z: 1.0
      w: 0.0
```

Looking at the code, I found [this part](https://github.com/ethz-asl/grid_map/blob/master/grid_map/src/GridMapRosConverter.cpp#L204-L207):

```
occupancyGrid.info.origin.orientation.x = 0.0;
occupancyGrid.info.origin.orientation.y = 0.0;
occupancyGrid.info.origin.orientation.z = 1.0;  // yes, this is correct.
occupancyGrid.info.origin.orientation.w = 0.0;
```

I don't understand why this change is necessary. Could someone please explain?

Looking at the **second issue** about the position offset the situation is similar. The original origin position in my example is:

```
    position: 
      x: -50.0
      y: -50.0
      z: 0.0
```

The position of the grid map's origin is the same. But during the conversion into a occupancy grid the origin is shifted/adjusted to:

```
    position: 
      x: 7.45058059692e-07
      y: 7.45058059692e-07
      z: 0.0
```

The reference frame for the origin of the occupancy grid in ROS (e.g. when published by `move_base`) seems to be one of the corners. When looking into the [conversion code ([part 1](https://github.com/ethz-asl/grid_map/blob/master/grid_map/src/GridMapRosConverter.cpp#L200), [part 2](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/src/GridMapMath.cpp#L161-L168)) it seems to be assumed that the reference frame is the centre. What's the reason behind this?

bit-pirate, 2016-02-10:
@stonier FYI

stonier, 2016-02-11:
First issue, I'm guessing its due to the data layout from the top left corner (graphics style). I haven't done a costmap <-> cost_map (aka grid_map) conversion yet, but will shortly so I'll probably experience the same thing.

Second issue, just conventions I think. Pinning the grid_map in the centre or bottom left corner is neither right, nor wrong, just different ways of doing it. So long as all is clearly defined, I'm ok with that. They have included a useful diagram referencing the bits and pieces of the grid_map layout [here](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/doc/grid_map_conventions.png).

bit-pirate, 2016-02-11:
Regarding the second issue, I think it's fine, if grid map uses it's own convention. Though when converting to occupancy grids, the output should align with the convention (mainly) used within ROS.

stonier, 2016-02-11:
I didn't seem to have any problems loading an image into a cost_map (aka grid_map) and then to occupancy grid map - setting the frame id to "map" and cost map position to zero resulted in an occupancy grid oriented in the expected x-direction in rviz. It is using the [same code](https://github.com/stonier/cost_map/blob/indigo-devel/cost_map/src/lib/converter.cpp#L302). Picture [here](https://github.com/stonier/cost_map/blob/indigo-devel/cost_map/doc/image_loading_coordinates.png).

Is your conversion from an occupancy grid into a grid_map doing things correctly?

bit-pirate, 2016-02-11:
I'm not changing any position or frame id, but use the source. Please post the pose/origin after your conversion from image (`sensor_msgs::Image`?) to cost_map/grid_map as well as the resulting occupancy grid message.

Here is my implementation of the conversion `nav_msgs::OccupancyGrid` -> `grid_map::GridMap`:

```c++

bool ElevatorController::occupancyGridToGridMap(const nav_msgs::OccupancyGrid& occupancy_grid_msg,
                                                grid_map::GridMap& grid_map)
{
  grid_map_msgs::GridMap grid_map_msg;
  grid_map_msg.info.header = occupancy_grid_msg.header;
  grid_map_msg.info.resolution = occupancy_grid_msg.info.resolution;
  grid_map_msg.info.length_x = occupancy_grid_msg.info.width * occupancy_grid_msg.info.resolution;
  grid_map_msg.info.length_y = occupancy_grid_msg.info.height * occupancy_grid_msg.info.resolution;
  grid_map_msg.info.pose = occupancy_grid_msg.info.origin;
  grid_map_msg.layers.resize(1);
  grid_map_msg.layers[0] = "occupancy_probabilities";
  grid_map_msg.basic_layers.resize(0);
  grid_map_msg.data.resize(1);
  grid_map_msg.data[0].layout.dim.resize(2);
  grid_map_msg.data[0].layout.dim[0].label = "column_index";
  grid_map_msg.data[0].layout.dim[0].size = occupancy_grid_msg.info.height;
  grid_map_msg.data[0].layout.dim[0].stride = occupancy_grid_msg.info.height * occupancy_grid_msg.info.width;
  grid_map_msg.data[0].layout.dim[1].label = "row_index";
  grid_map_msg.data[0].layout.dim[1].size = occupancy_grid_msg.info.width;
  grid_map_msg.data[0].layout.dim[1].stride = occupancy_grid_msg.info.width;
  grid_map_msg.data[0].layout.data_offset = 0;
  grid_map_msg.data[0].data.resize(occupancy_grid_msg.data.size());
  for (int cell = 0; cell < occupancy_grid_msg.data.size(); ++cell)
  {
    grid_map_msg.data[0].data[cell] = occupancy_grid_msg.data[cell];
  }
  grid_map_msg.outer_start_index = 0;
  grid_map_msg.inner_start_index = 0;

  if (grid_map::GridMapRosConverter::fromMessage(grid_map_msg, grid_map))
  {
#ifdef DEBUG
      pub_gridmap_.publish(grid_map_msg);
      nav_msgs::OccupancyGrid occ_grid_msg;
      grid_map::GridMapRosConverter::toOccupancyGrid(grid_map, grid_map.getLayers()[0], 0, 100, occ_grid_msg);
      pub_costmap_.publish(occ_grid_msg);
    }
#endif
    return true;
  }
  else
  {
    ROS_WARN_STREAM_THROTTLE(1.0,  "Couldn't convert grid map message into grid map.");
    return false;
  }
};
``

czalidis, 2016-02-11:
@bit-pirate You can **not** just copy the origin from the OccupancyGrid to the GridMap, you have to convert it first. In the OccupancyGrid msg the origin represents the real-word pose of the cell (0, 0), while GridMap's position is w.r.t. the map center, as you can see at the layout that @stonier mentioned.
See here: http://docs.ros.org/api/nav_msgs/html/msg/MapMetaData.html

pfankhauser, 2016-02-11:
Thanks a lot, @bit-pirate, @stonier, and @czalidis. I just added the conversion from occupancy map to grid map in the branch `feature/occupancy_to_grid_map`. Could you please test this for your use case? If all works out, I will merged this branch into master. If you have any unexpected behavior, please document it with a unit test which you can create in `GridMapTest.cpp`.

pfankhauser, 2016-02-12:
This issue should is resolved with #36.

closed 2016-02-12 by pfankhauser

bit-pirate, 2016-02-15:
Was fighting my own installation on Friday and finally got around testing your changes just now.

Works well for me. :+1: Thanks for this addition!