ANYbotics/grid_map #34: Release with metapackage

bit-pirate, 2016-02-11:
I find the ROS wiki package documentation quite useful when learning about new code. So far only [`grid_map`](http://wiki.ros.org/grid_map) had a page and so I have started adding empty wiki pages for the other packages as well, e.g. http://wiki.ros.org/grid_map_core).

Now, each package looks quite lonely, since there is no stack/metapackage linking them. Hence, I like to suggest adding a metapackage, which links to all grid_map packages. Doing so will enable the ROS wiki to link between the packages via a stack page. This page is also well suited to be a landing page from which one can point to the other important packages.

Also installing the metapackage will automatically pull in the other packages.

An example is the [rocon metapackage](https://github.com/robotics-in-concert/rocon/tree/indigo) and the related [wiki page](http://wiki.ros.org/rocon) (notice the list of packages below the release buttons).

pfankhauser, 2016-02-11:
Thanks for the info, a meta package certainly makes sense. Do you suggest me to create a new `grid_map_meta` package, or rename the current `grid_map` package to `grid_map_ros` and use the `grid_map` package as meta package?

bit-pirate, 2016-02-12:
I personally like the latter better - logically and it seems also a de-facto standard within ROS - but it's also more work. If you don't mind, go for it. :-)

Metapackage name alternatives: `grid_map_all`, `grid_map_full`

Maybe @stonier has more ideas?

stonier, 2016-02-12:
Whatever works :neckbeard:

pfankhauser, 2016-03-02:
This is resolved in #44.

closed 2016-03-02 by pfankhauser