ANYbotics/grid_map #45: Iterating w/ speed in mind

stonier, 2016-03-02:
In the [iterators demo](https://github.com/ethz-asl/grid_map/blob/master/grid_map_demos/src/IteratorsDemo.cpp#L54), the iteration is using a string key based lookup to the layers to pull data at every index, which can be slow if you're just wanting to do full iterations over the data.

My first take was just to pull the data matrices for the layers and iterate over them directly, however after understanding a bit better what is happening after the `getPosition` discussion, I gather this is a bad idea since a move will have shifted your sense of where the start is in that data matrix.

So instead, I've taken to really using the grid map iterators to pull the indices and operate on the matrix directly as follows:

```
grid_map::Matrix& data_layer_one = my_grid_map.get("layer_one");
grid_map::Matrix& data_layer_two = my_grid_map.get("layer_two");
for (grid_map::GridMapIterator iterator(my_grid_map); !iterator.isPastEnd(); ++iterator) {
  int i = (*iterator)(0);
  int j = (*iterator)(1);
  // do something with data_layer_one(i, j);
  // do something with data_layer_two(i, j);
}
```

Is this the recommended way? If it is, it would be good to have that in the iterators demo to show how you might iterate over a grid map with speed.

pfankhauser, 2016-03-03:
Thanks for your comments. You are right that the performance can be increased when locally storing a reference or copy of the data layer. There's actually a node that runs a performance comparison:

    rosrun grid_map_demos iterator_benchmark

In this example, we iterate through a grid map and copy the data from one layer into another if it is bigger in value. For a map of size 5000 x 5000 cells (25 mio. cells), the performance can be improved from the convenient usage of the iterator to the optimized usage from 1350 ms to 135 ms.

I have updated the documentation to highlight this here: https://github.com/ethz-asl/grid_map#iterators

Does this clarify your question?

pfankhauser, 2016-03-03:
Here's also an updated version of your example for the fastest execution that I have been able to achieve yet:

    grid_map::Matrix& data_layer_one = my_grid_map.get("layer_one");
    grid_map::Matrix& data_layer_two = my_grid_map.get("layer_two");
    for (grid_map::GridMapIterator iterator(my_grid_map); !iterator.isPastEnd(); ++iterator) {
      const int i = iterator.getLinearIndex();
      // do something with data_layer_one(i);
      // do something with data_layer_two(i);
    }

Here we make use of the linear index access of Eigen matrices to maximize performance.

stonier, 2016-03-03:
> Does this clarify your question?

Yes, and thanks for the pointer about the getLinearIndex().

You might like to add a comment to the iterators demo program - many programmers (myself included) tend to go directly to code before documents to discover how to do something specific.

pfankhauser, 2016-03-03:
Yes, good idea, thanks.

closed 2016-03-03 by pfankhauser