ANYbotics/grid_map #53: Submap has different size than given in parameters provided to getSubmap()

skohlbr, 2016-03-24:
I've been running into a problem with maps not matching up that I suspect could be related to mismatching sizes. So what I have is the following: A ```static_map_``` containing standard occupancy data that is fixed wrt to world. I also have a "moving" elevation map ```local_elevation_map_```. I want to fuse the local elevation data to ```static_map_```, adding obstacles based on elevation data.

To this end, I do the following:

  ```
  const grid_map::Length& local_length = local_elevation_map_.getLength();
  const grid_map::Position& local_position = local_elevation_map_.getPosition();

  static_map_.extendToInclude(local_elevation_map_);

  bool submap_create_success;
  grid_map::GridMap static_cut = this->static_map_.getSubmap(local_position, local_length, submap_create_success);

  grid_map::Matrix& elev_data   = local_elevation_map_["elevation"];
  grid_map::Matrix& static_cut_data = static_cut["occupancy"];

  for (grid_map::GridMapIterator iterator(local_elevation_map_); !iterator.isPastEnd(); ++iterator) {
      const grid_map::Index index(*iterator);

      // According to some rule, add obstacle information to the map
      if ( std::abs( robot_elevation - elev_data(index(0), index(1)) ) > 0.38 ){
         static_cut_data(index(0), index(1)) = 100.0;
      }
  }
  static_map_.addDataFrom(static_cut, true, true, true);
```

So essentially, my goal was to create a subMap with exactly the same size and origin as the local elevation map and then iterate over/compare them using the same index. This works in the sense that I get data out, but it looks transformed incorrectly (i.e. obstacles in the elevation map show up in a different place in the ```static_map_``` when visualizing them). With some debugging I found that the length of my ```local_elevation_map_``` is 6, but that of the generated submap ```static_cut``` is 6.05. Now I can see how this can happen due to discretization, but it's highly inconvenient :)

I thus have a  request and a question:

It would probably be good to indicate in documentation that the actual size of the returned submap might be different from what was requested. Alternatively, if possible it could be made to actually always have the requested size, but I think this might be tricky due to the discrete nature.

What would be the "proper" way of doing what I want (add data from a local map to a world fixed one based on some rule)?

skohlbr, 2016-03-24:
Ok, so I managed to make it work using

```
  for (grid_map::GridMapIterator iterator(static_cut); !iterator.isPastEnd(); ++iterator) {
      const grid_map::Index index(*iterator);

      grid_map::Position position;
      static_cut.getPosition(index, position);

      grid_map::Index elev_index;
      local_elevation_map_.getIndex(position, elev_index);

        if ( std::abs( robot_elevation - elev_data(elev_index(0), elev_index(1)) ) > 0.38 ){
          static_cut_data(index(0), index(1)) = 100.0;
        }
  }
```

I'd still be interested in a more elegant/efficient way of doing this however.

pfankhauser, 2016-03-24:
Assuming you have the static and the local maps with following layers
```
GridMap static_map_({"occupancy"});
GridMap local_elevation_map_({"elevation", "occupancy"});
```
with different positions, sizes, and resolutions (hence, the grids are not aligned). Instead of the `for`-loop, you can achieve the same directly with:
```
local_elevation_map_["occupancy"] = ((local_elevation_map_["elevation"].array() - robot_elevation).abs() > 0.38).cast<float>() * 100.0;
static_map_.addDataFrom(local_elevation_map_, true, true, false, {"occupancy"});
```

Good point about the documentation, I added a comment in e93d7aa38db9fa4cb394c70746a110eb8e050409. Thanks!

closed 2016-03-24 by pfankhauser

skohlbr, 2016-03-24:
Thanks for the quick answer, that indeed looks a good bit more elegant and efficient :)