ANYbotics/grid_map #56: Index use inconsistent between toCvImage(..) and addLayerFromImage(..)

skohlbr, 2016-03-28:
In ```toCvImage``` the order of the index access (1,0) into the ```cv::Mat``` ([code](https://github.com/tu-darmstadt-ros-pkg/grid_map/blob/master/grid_map_ros/src/GridMapRosConverter.cpp#L473)) is exactly in opposite order to access in ```addLayerFromImage``` (0,1) ([code](https://github.com/tu-darmstadt-ros-pkg/grid_map/blob/master/grid_map_ros/src/GridMapRosConverter.cpp#L397)).

This means that when creating a ```cv::Mat``` from a grid map layer, performing some operations on it and then adding it back as another layer using both conversion methods mentioned above, the new layer will be rotated by 90 degrees. Looks like a bug to me, or is there some kind of rationale for this?

skohlbr, 2016-03-28:
Ok, so getting after getting crashes with non-square ```cv::Mat``` images generated using ```toCvImage``` (and seeing invalid writes in valgrind) I now think this actually has to be fixed in ```toCvImage``` instead of ```addLayerFromImage()```. See #58.

pfankhauser, 2016-03-29:
Thanks a lot for your investigations, Stefan. I'm currently working on a round trip unit test for the image conversion. I'll let you know once I have more information.

pfankhauser, 2016-04-14:
I have completely rewritten the to image conversions from and to OpenCV images:
https://github.com/ethz-asl/grid_map/tree/fix/image_conversion

There's now a new package `grid_map_cv` which can be used for working with OpenCV on grid maps and is ROS independent. Also, conversions are much faster now.

Could you please check if the methods in this branch are working for your application?

PR #61 (2016-04-15): grid_map_cv
Complete rewrite the to image conversions from and to OpenCV images. There's now a new package `grid_map_cv` which can be used for working with OpenCV on grid maps and is ROS independent. Also, conversions are much faster now and fixes #56.
closed 2016-04-15 by pfankhauser