ANYbotics/grid_map #60: Changing the resolution of a grid_map

stonier, 2016-04-07:
Without actually changing the lengths, nor the position of the grid_map window.

We've been playing around with something in [cost_map/pull-requests/17](https://github.com/stonier/cost_map/pull/17).

Thoughts?

pfankhauser, 2016-04-15:
Here are two easy ways to change the resolution of a grid map.
* Create a new grid map with the desired resolution, and use the `addDataFrom(...)` from method to add the data from the source grid map.
* Use the `changeResolution(...)` method of the new `GridMapCvProcessing` class in *grid_map_cv*. This methods leverages the OpenCV interpolation functions to change the resolution. You can run the demo with

        roslaunch grid_map_demos resolution_change_demo.launch

Does this solve your problem?

pfankhauser, 2016-05-13:
@stonier Any update on this?

closed 2016-09-13 by pfankhauser

stonier, 2016-09-21:
Missed my emails on this one. `addDataFrom()` could work, though it means manually doing the rest of the details (setting the length, position) manually. I feel having methods separate to the class (like the `changeResolution()` is the best course here since there are multiple ways of handling the changed resolution.

For our needs, I probably will build our own `changeResolution()` methods. Linking in `openCV` is often undesirable for us - we often drop it for product code.

pfankhauser, 2016-09-21:
Ok, I see. In this case, I think the best approach would be to create a new class `GridMapProcessing` in `grid_map_core` to perform a resolution change without dependency on OpenCV. Let me know if you need any support.

BrunoEduardoCSantos, 2018-11-27:
@pfankhauser I added data another grid using `addDataFrom()`  and publish as occupancy grid. Although, for some reason the occupancy grid is swinging around and it doesn't always align with grid map . Any thoughts?
Thanks

BrunoEduardoCSantos, 2018-11-27:
PS: I am using the following flags in `addDataFrom()`   :
`new_grid_map.addDataFrom(newGridMap,true,false,false, layers)`