ANYbotics/grid_map #66: iterator_demo throws exception

solosuper, 2016-08-25:
When running the iterator demo, the program terminates after a while with the following message:

```bash
$ roslaunch grid_map_demos iterators_demo.launch
# ...
[ INFO] [1472111866.193859810]: Running polygon iterator demo.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
```

pfankhauser, 2016-08-26:
The demo runs through several iterators and should terminate after the last iterator. You should see something like this

    [ INFO] [1472222564.624084658]: Running polygon iterator demo.
    [grid_map_iterators_demo-2] process has finished cleanly

How long does the demo run for you? Can you run GDB on it?

solosuper, 2016-09-13:
Hi, sorry for the late reply.
I should probably mention that I have cloned this repo into my workspace and is thus compiling it from source with catkin_make along with my own package (and cv_bridge (to be able to use OpenCV 3)). I just tried installing grid_map with `apt-get install ros-indigo-grid-map` and the demo then works perfectly.

--- 

So to the problem:

Yes, I've tried running GDB on it, and the demo stops during the polygon iterator demo, at [IteratorsDemo.cpp#L195]

[IteratorsDemo.cpp#L195]: https://github.com/ethz-asl/grid_map/blob/master/grid_map_demos/src/IteratorsDemo.cpp#L195

solosuper, 2016-09-13:
I am embarassed to say that the issue should not have been reported. I was working on an old fork of this project, and I guess that's the cause of error.

closed 2016-09-13 by solosuper

pfankhauser, 2016-09-13:
Don't worry, thanks for the update.