ANYbotics/grid_map #67: Eigen Plugins Order

pfankhauser, 2016-08-26:
Currently, there's an error when Eigen is not loaded at first from `grid_map_core` because of the Eigen Plugins.

solosuper, 2016-10-12:
When I am using `"#include "grid_map_cv/grid_map_cv.hpp"`, I get the error 

```bash
/opt/ros/kinetic/include/grid_map_cv/GridMapCvConverter.hpp: In static member function ‘static bool grid_map::GridMapCvConverter::toImage(const grid_map::GridMap&, const string&, int, cv::Mat&)’:
/opt/ros/kinetic/include/grid_map_cv/GridMapCvConverter.hpp:136:47: error: ‘const Matrix {aka const class Eigen::Matrix<float, -1, -1>}’ has no member named ‘minCoeffOfFinites’
     const float minValue = gridMap.get(layer).minCoeffOfFinites();
```
Has it got something to do with this issue, or how can I resolve it?

pfankhauser, 2016-10-20:
Yes, that's the problem we're facing when Eigen is included first, as Grid Map relies on Eigen Addons that are loaded through macros: http://eigen.tuxfamily.org/dox/TopicCustomizingEigen.html

I'm not sure how to resolve this nicely, any hints are appreciated.

pfankhauser, 2016-10-21:
Thanks to @djud and @hogabrie for resolving this with https://github.com/ethz-asl/grid_map/pull/73!

closed 2016-10-21 by pfankhauser

adricort, 2024-05-16:
Hi @pfankhauser the pull is no longer available and I still have the same problem, do you know how to solve this? thanks