ANYbotics/grid_map #75: addLayerFromImage broken (?) for cv::Mat images of float type

skohlbr, 2016-11-03:
I tried converting a floating point `cv::Mat` to a layer and noted that the grid map ends up having extremely small entries (magniture ~E-37). The cv image however has entries in the range [0, 31.0]. Investigating I found the following line: https://github.com/ethz-asl/grid_map/blob/master/grid_map_cv/include/grid_map_cv/GridMapCvConverter.hpp#L96
Setting the max by casting to `(float)std::numeric_limits<Type_>::max();` works for small integer types, but  not if `Type` is a float (or double).

I'll do my own small conversion for the moment, but two action items might be useful:
* Document that float images are not really supported by the function
* Possibly add a converter that can directly convert float cv::Mat to GridMap without scaling the entries

pfankhauser, 2016-11-03:
Thanks for reporting. Can you check if branch [fix/image_float_conversion](https://github.com/ethz-asl/grid_map/tree/fix/image_float_conversion) solves this issue for you?

pfankhauser, 2016-11-11:
@skohlbr Any news on this?

skohlbr, 2016-11-11:
After looking into this some more I noticed that mapping the data pointer can be directly used to initialize a cv::Mat without copying, so I switched to that approach (see [here](https://github.com/tu-darmstadt-ros-pkg/grid_map_proc/blob/master/src/grid_map_transforms.cpp#L18)). If you insist, I can try recreating my prior setup :)

pfankhauser, 2016-11-24:
Ok great, yes that works too if one is careful ;-)

closed 2016-11-24 by pfankhauser