ANYbotics/grid_map #76: Conversion from CostMap2D to GridMap

skohlbr, 2016-11-11:
Standard ROS navigation uses [CostMap2D](https://github.com/ros-planning/navigation/blob/0fec250ead4a8b2c5ed2e6d1e3efb28e7885c12e/costmap_2d/src/costmap_2d.cpp) as a cost map representation and when implementing a [nav_core::BaseGlobalPlanner](https://github.com/ros-planning/navigation/blob/0fec250ead4a8b2c5ed2e6d1e3efb28e7885c12e/nav_core/include/nav_core/base_global_planner.h#L48) plugin that uses grid_map as a back-end, it would be desirable to have a converter that allows initalizing the internal grid_map from a CostMap2D.
Are there any plans for implementing this (or even some implementation out there that I missed)?

stonier, 2016-11-21:
We have a function in the `cost_map` library which does this...[here](https://github.com/stonier/cost_map/blob/devel/cost_map/src/lib/converter.cpp#L347). There's an active [PR](https://github.com/stonier/cost_map/pull/19) on that too looking to improve it a bit.

At any rate, it could very easily be templatised to handle `grid_map` and `cost_map` objects together I think.

skohlbr, 2016-11-23:
Thanks for the link @stonier! I made my own small converter by pulling out some code from navigation, converting from CostMap2D to nav_msgs/OccupancyGrid and then just using the conversion to GridMap. Not terribly efficient, but gets the job done: https://github.com/tu-darmstadt-ros-pkg/grid_map_navigation_planner/blob/master/grid_map_planner_nav_plugin/include/grid_map_planner_nav_plugin/grid_map_cost_map_2d_conversion.h#L71

stonier, 2016-11-23:
Oh, two step. Bit expensive, but it should work.

Good to have this issue here. I've been a bit delayed getting my cost map revamped (has to wait until I get back from interview travels), but this would be a good function to get into the core for both cost_map and grid_map.

pfankhauser, 2016-11-24:
Thanks @skohlbr and @stonier for your input! I added a quick implementation of a Costmap2D to Grid Map in the [feature/costmap_conversion](https://github.com/ethz-asl/grid_map/tree/feature/costmap_conversion) branch.

- The class `Costmap2DConverter` provides two methods
    1. First one has to initialize a new grid map with  `initializeFromCostmap2d` to adapt to the geometry of the costmap.
    2. Then, data from the costmap can be copied to a layer of the grid map with `addLayerFromCostmap2d`.
- Direct conversion from Costmap2d to Grid Map, no intermediate steps.
- Templated on `MapType`, so can be used by Grid Map derivatives as well.
- Unit tests [here](https://github.com/ethz-asl/grid_map/blob/feature/costmap_conversion/grid_map_ros/test/Costmap2DConverterTest.cpp).

Please let me know if this works for you or if you have any suggestions. Thanks!

pfankhauser, 2016-11-24:
Update: I used the cost translation table as suggested by @skohlbr and added him as an author, please let me know if this is not ok.

skohlbr, 2016-11-24:
Fine with me, but full disclosure: I extracted and modified that code from [costmap_2d_publisher](https://github.com/ros-planning/navigation/blob/53087ca5242fb40f06c84a6bcc74057c3470eff7/costmap_2d/src/costmap_2d_publisher.cpp) (the cost translation table is a `static char* cost_translation_table_` there).

pfankhauser, 2016-11-29:
@skohlbr Ok great, were you able to check if the new conversion feature in branch [feature/costmap_conversion](https://github.com/ethz-asl/grid_map/tree/feature/costmap_conversion) works for your use case?

skohlbr, 2016-12-02:
Not yet, but will do next week.

stonier, 2016-12-27:
We've been refining what we do with Costmap2DROS conversions into cost maps - details [here](https://github.com/stonier/cost_map#costmap2dros-conversions). Basically the use case we were interested in supporting is the ability to catch a subwindow of the cost map around the ros pose.

This required us to be a little more careful (i.e. catch a few bugs) to make sure we don't introduce errors by having the resulting cost map not exactly aligning with the underlying ros costmap - it is not as straight forward as just setting the cost map centre to the robot pose position.

Anyway, seems to be working now and we have a battery of tests catching the different permutations. @pfankhauser it looks like something I could easily template in the style you made in the PR so we could use it for both cost map and grid map converters, just with different policies applied (e.g. direct value cast, or normalising from 0 to 100.0 or similar).

pfankhauser, 2017-01-24:
Solved by #78.

closed 2017-01-24 by pfankhauser