ANYbotics/grid_map #85: Building local grid map with stereo camera

montecarlo13, 2017-01-12:
hello guys, I want to build my own grid map on ROS Indigo with a sensor (e.g. kinect or Realsense camera) but without using a robot and navigation, just building the local grid map that should represent the obstacles detected by the sensor. What should I do with this grid map library? Do I need to work with the node "elevation mapping" in addition? Thanks!

pfankhauser, 2017-01-13:
Hi @montecarlo13 
In that case you can either use [Elevation Mapping](https://github.com/ethz-asl/elevation_mapping) or add the point clouds to a grid map in your own node, both ways will work.
Does that answer your question?

montecarlo13, 2017-01-13:
Hi @pfankhauser 
Thank you very much for your answer. Now I have a idea, I would firstly use OpenCV to process my depth images, which published by the sensor, then in my node I use the classes GridMapCvConverter and GridMapCvProcessing from the grid map library for converting the cv messages to /grid_map_msgs/GridMap, so that I can get the OccupancyGrid map with help from the node grid_map_visualization. Will that work? Thanks!

pfankhauser, 2017-01-24:
I don't think you need OpenCV for this. Just take your depth images, turn them into point clouds and either use Elevation Mapping or your own implementation to add the point clouds to a grid map. Does this help?

montecarlo13, 2017-01-28:
@pfankhauser

Thank you very much, it's really helpful ! :-)

closed 2017-02-07 by pfankhauser