ANYbotics/grid_map #89: [grid_map_cv] Handle NaN values in toImage

romainreignier, 2017-02-22:
When converting a GridMap to an OpenCV image with the method `grid_map:GridMapCvConverter::toImage`, it seems that the `NaN` values are converted to the value of the argument `lowerValue` by the `Clamp` function.

Maybe I am not using the library in the correct way so please, correct me if I am doing something wrong.

I am using a map during a SLAM so I get an OccupancyGrid with the value -1 for a cell not yet discovered (unknown).

1. I convert it to a GridMap so the unknown cells have the value `NaN`
2. I convert the GridMap to an OpenCV image with 100 as `lowerValue` and 0 as `upperValue` so I have black cells for the obstacles and white cells for the free space. But `NaN` values get automatically translated in white pixels so I lose the information about unknown cells.

For now, my workaround is to manually convert `NaN` cells to the value 50 but I thought maybe it could be handled by the library.

pfankhauser, 2017-02-24:
Are you using an image with a channel for the alpha layer? There's a [round trip unit test](https://github.com/ethz-asl/grid_map/blob/master/grid_map_cv/test/GridMapCvTest.cpp#L47) that passes for the `CV_8UC4` CV image type and I just added a nan-value/transparent pixel [here](1d83ba6fdc7782b69ac7400b8a7f09531251b49e).

If this does not solve your problem, could you please create a PR with a failing unit tests that shows the issue?

romainreignier, 2017-02-24:
Ah no, I forgot to mention that I am using a `CV_8UC1` grayscale image, that is why, there is no other solution to translate a `NaN` to a mid value between Free and Occupied.

pfankhauser, 2017-03-02:
So is there anything you would like to see differently in the behavior of Grid Map or should I close this issue?

romainreignier, 2017-03-02:
What do you think about setting `NaN` values to the mid value between `lowerValue` and `upperValue`?

romainreignier, 2017-03-02:
*if there is no alpha channel, of couse.

pfankhauser, 2017-03-02:
I would keep the current behavior, since I feel it's the one with the least assumptions. Your use case is very special, and if I'm not mistaken, you could achieve your desired behavior by setting your empty/nan cells to the mid-value _before_ converting it to the image format.

romainreignier, 2017-03-02:
Yes, of course, it is what I have done.
But for the `map_server` for example, a unknown cell is a pixel with value 128 that is why I was surprised that `grid_map` did not picked a middle value.
Anyway, as you said, my use case may be special for the `gird_map` because I did not managed to find an easy way to convert pixels position to `grid_map` coordinates then `OccupancyGrid` coordinates. So I have ended up writing my own conversion methods.
Maybe you could add in the doc that the `NaN` values are converted to the value 0 for images with no alpha layer.

JuanLuisHortelano, 2017-06-20:
I have also encountered this problem.

The issue is that the output of SLAM algorithms for unknown area is -1 in occupancy grids, that gets transformed into NaN in Grid maps that gets transformed into 0 in cv::Mat. That causes cv::Mats to make no difference between occupied cells and unknown cells.

I don't think changing NaN cells to a middle value before converting to grid map is a good solution because an occupancy grid is supposed to give a probability of occupancy of each cell and is not supposed to give a value to unknown space.