ANYbotics/grid_map #90: grid_map doesn't transform according to pose in message

mauricefallon, 2017-02-24:
As far as I can tell grid_map cannot transform according to the z value or orientation in GridMapInfo.msg
* [doesn't use 3D values in the message](https://github.com/ethz-asl/grid_map/blob/master/grid_map_ros/src/GridMapRosConverter.cpp#L42)
* [setGeometry accepts only x and y](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/include/grid_map_core/GridMap.hpp#L68)
It only uses the x,y value

We have our own heightmapping tool which creates a grid relative to a frame.
[image]
... note that the large grid in the 'director' UI is not aligned with the heightmap grid (which is aligned with the robot)

I suppose the solution would be to add our frame to TF and set the x,y values to zero. But I wanted to ask why a full 3d pose is included in the ROS message but not used?

pfankhauser, 2017-02-24:
Maurice, yes, currently only _x_ and _y_-positions are supported. It would be easy to add _z_-position support, but rotations would increase the complexity considering that Grid Map supports additionally to `setPosition(...)` the `move(...)` command (see [here](https://github.com/ethz-asl/grid_map#changing-the-position-of-the-map)).

I will look into better documenting this and/or adapting the ROS message to clarify this.

Is this only a visualization issue? If yes, then you can easily get by with introducing an additional TF frame.

mauricefallon, 2017-02-24:
Yes, I think adding a TF frame is the solution for me.

From what I see, the pose in the GridMap message is a kind of convenience. (I think you use it to support GridMaps with variable resolution). It wasn't clear what its purpose was.

closed 2017-02-24 by mauricefallon