ANYbotics/grid_map #98: LineIterator has potential bug?

victl, 2017-05-21:
I've been using grid_map for a while. I found LineIterator occasionally crash if I initialize it with two 'Position's, and access the grid map with
```
LineIterator iter(*grid_map_, pos1, pos2);
for (; !iter.isPastEnd(); ++iter)
{
  grid_map_->at("map", *iter) = foo;
}
```
This is most likely when both 'pos1' and 'pos2' are out of valid range.

I looked into the source code of 'LineIterator' and found that the constructor will NOT initialize anything if 'getIndexLimitedToMapRange()' returns false. In this case, when you invoke 'isPastEnd()', it'll probably return false because it actually returns 'iCell_ > nCells_'. NOTE the 'iCell_' and 'nCells_' were not initialized properly.

pfankhauser, 2017-05-24:
Thanks for reporting this! We have a [unit test](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/test/LineIteratorTest.cpp#L74) for the case where both start and end point are outside the map, so I'm unsure what happens in your case.

Could I ask you to recreate the erroneous behavior from your description in a unit test and create a PR?

jinkunw, 2018-12-07:
I met the same problem. When both `getIndexLimitedToMapRange` failed, `iCell_` and `nCells_` were not initialized. The problem is solved if I simply set them to zero in the constructor.

mkhayatian, 2023-02-02:
The problem with current implementation is that if the start and end points are outside of the gridmap size and the (start,end) line doesn't intersect with the grid_map, it will `throw std::invalid_argument("Failed to construct LineIterator.");` 
(line 24 of `LineIterator.cpp`). This is because the `getIndexLimitedToMapRange()` function starts from start points and moves towards the end point and if no point inside the gridmap is found (index), then it will return false.

mkhayatian, 2023-02-02:
Also, a better implementation for `getIndexLimitedToMapRange()` is finding the intersection between two lines (start,end) and (gridmapEdge[i], gridmapEdge[i+1]) instead of a while loop.