ANYbotics/grid_map #102: access normals from gridmap variable

mcamurri, 2017-06-08:
I have a question: from the RViz I can see that the gridmap is visualized with some sort of normal to orient the cell/mesh. This means that for each cell a normal is computed. Is it available from the gridmap object or it is computed by the RViz plugin from the neighbors?

Is there a method to get the normals automatically?

pfankhauser, 2017-06-09:
Hi @mcamurri,
We have implemented surface normal estimation as part of [Elevation Mapping](https://github.com/ethz-asl/elevation_mapping) and as a Grid Map Filter in [Traversability Estimation](https://github.com/ethz-asl/traversability_estimation). We are planning to unify the implementation as a Grid Map Filter in this repo, but in the meanwhile you can use the other implementations.

closed 2017-06-09 by pfankhauser

pfankhauser, 2017-08-23:
It's here! We have a new implementation for several grid map filters, including normal vector estimation. More info [here](https://github.com/ethz-asl/grid_map/blob/master/README.md#grid_map_filters).

[image]