ANYbotics/grid_map #106: PolygonIterator getting stuck

KlausBuchegger, 2017-06-27:
Hi, I can't 100% reproduce this bug, but it seems to me that if one (or more?) of the vertices of the polygon used for the PolygonIterator is outside of the map, for some reason my code just hangs.

It does not happen all the time when I have a polygon that's not fully inside the map, but it happens a lot, and if i reexecute some scenario that got stuck it always gets stuck again.

Anyone else ever experienced something like this?

pfankhauser, 2017-07-06:
Hi @KlausBuchegger! I'm sorry you're experiencing problems with Grid Map. Can you help me find and solve this bug? It would be really helpful if you could recreate your scenario in a [unit test](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/test/PolygonIteratorTest.cpp) and if necessary randomize the values and repeat the test until the error occurs?

KlausBuchegger, 2017-07-06:
Sure, I'll try and do that if I find some time, hopefully early next week.

KlausBuchegger, 2017-07-11:
So, I tried but failed reproducing my scenario where I got stuck (don't have the exact code anymore).
However, during my experiments I noticed that the GridMap function 
getPosition (const Index &index, Position &position)
fails for an Index with negative x or y values.
For example I have a 5x5m map with resolution 0.1 and Position(0,0) as center.
a) Then Index(0,0) ~ Position(2.45,2.45)
b) Index(10,1) ~ Position(1.45,2.35) so far so good
But:
c) Index(-10,1) ~ Position(-1.45,2.45)
d) Index(0,-3) ~ Position(2.3342e-312,0)
I'm not sure what the expected behavior for a negative index should be, my intuition was it would just correspond to a Position outside of the map?
But the y coordinate of c) should definitely be 2.35 to match b) and not 2.45, right?

KlausBuchegger, 2017-09-09:
Any progress on this issue?