ANYbotics/grid_map #111: Moving Costmap2DConverter.hpp breaks downstream packages

tfoote, 2017-07-25:
The recent moving of the include file: https://github.com/ethz-asl/grid_map/commit/787747e568edc2f05c1a1d7841e0682424087507 in #96 is breaking downstream packages expecting the header.

A backwards compatibility header with a deprecation warning would be appreciated.

http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__cost_map_ros__ubuntu_trusty_amd64__binary/8/console
```
00:07:47.321 /tmp/binarydeb/ros-indigo-cost-map-ros-0.3.1/src/lib/converter.cpp:23:47: fatal error: grid_map_ros/Costmap2DConverter.hpp: No such file or directory
00:07:47.321  #include <grid_map_ros/Costmap2DConverter.hpp>
00:07:47.321    
```

pfankhauser, 2017-07-25:
Thanks for the notice @tfoote! I just added the backward compatibility in https://github.com/ethz-asl/grid_map/pull/112. I'll merge this and create a new version with the hope to fix things asap. Let me know if there's anything else I can do.

stonier, 2017-07-25:
Peter, I can update cost_map to accomodate the moved header on the weekend.

tfoote, 2017-07-25:
Thanks for the quick fix @pfankhauser it looks like this is fixing the regressions. Once the rebuild is complete I think we'll be ready to sync.

closed 2017-07-25 by tfoote

pfankhauser, 2017-07-25:
Ok great, @tfoote!

@stonier: yes, please update costmap_2d such that in the next version I can remove the deprecated header. Thanks!

stonier, 2017-08-08:
@pfankhauser `cost_map` is updated for both indigo and kinetic (no lunar version yet).

pfankhauser, 2017-08-15:
Ok great, thanks @stonier!