ANYbotics/grid_map #132: No possibility to `resize` a GridMap ?

artivis, 2017-09-24:
Hi,
I started using the `grid_map_core` pkg recently aiming at incrementally building a `GridMap`. However it seems there are no possibility to resize 'on the fly'  a `GridMap` (resize while preserving the existing data).   
The `setGeometry` functions internally call `clearAll` thus aren't suited for incremental building. I came across both `addDataFrom` & `extendToInclude` functions which both require as input argument another `GridMap` which is a bit of an overshoot - instantiation of a second `GridMap` object, internal copy of the current object etc...  
Did I miss something or is it a missing functionality ??

pfankhauser, 2017-09-25:
Hi @artivis, thanks for sharing! I agree, a resizing method which is easier to use then `extendToInclude` could be implemented. Internally, the `extendToInclude` could call this resizing method. How would you like to use the `resize` method? Would you give it a new size and and leave the position the same, or something else?

artivis, 2017-09-25:
I am not familiar enough with the internal of `grid_map_core` to fully grasp the difficulty of implementing such new feature. But IMHO such resizing method would allows the user for either appending new cells on either side of the `GridMap` or both sides - e.g. increasing width lead to appending new cells both on the 'left' and 'right' of the `GridMap` and similarly for increasing the height. Increasing the full size would be something like 're-centering' the existing data on a larger grid. The desired behavior could be defined with an input argument such as an `enum`.

laxnpander, 2017-12-01:
Is there any update on this? Could be useful for my use-case too. An incremental resize without clearing all data sounds like a good thing to do. I personally won't need to reposition the map.

zhixuan-wei, 2018-01-25:
Add a `resize` method is a good idea! It's better to set both size and position when resize (move and resize at the same time). More over, is it possible to set the initial value of the new cells? Thank you!