ANYbotics/grid_map #135: Compilation Error in Indigo 1.11.20

hujun1413, 2017-10-10:
When I run `catkin_make`, I got following:
```
ExpressionFilter.cpp:56:88: error: no matching function for call to ‘grid_map::SlidingWindowMathExpressionFilter<grid_map::GridMap>::getParam(std::string, bool&)’
if (!FilterBase<T>::getParam(std::string("compute_empty_cells"), isComputeEmptyCells_)) {
                                                                                        ^
/home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_filters/src/SlidingWindowMathExpressionFilter.cpp:56:88: note: candidates are:
In file included from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_filters/include/grid_map_filters/SlidingWindowMathExpressionFilter.hpp:15:0,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_filters/src/SlidingWindowMathExpressionFilter.cpp:9:
opt/ros/indigo/include/filters/filter_base.h:121:8: note: bool filters::FilterBase<T>::getParam(const string&, std::string&) [with T = grid_map::GridMap; std::string = std::basic_string<char>]
   bool getParam(const std::string& name, std::string& value)
...
```

It seems like the filter_base.h file in Indigo 1.11.20 have no override function for getParam(string&, bool ).
Also I found filter_base.h has getParam(string&, bool ) in Indigo 1.11.21 on my friend's computer, so should I **build this pakage in higher Indigo version like 1.11.21**?

pfankhauser, 2017-10-10:
Yes, you should update to the newest version of Indigo. Alternatively, you can clone an older release (e.g. https://github.com/ethz-asl/grid_map/releases/tag/1.5.2) which does not include the latest changes for grid map filters. Let me know if this resolve your issue.

hujun1413, 2017-10-11:
I clone the older release 1.5.2，and it works on Indigo 1.11.20!

Thank you for your answer, later I'll try the newest grid_map version on Indigo 1.11.21. 

By the way, do you know how to update Indigo version? I found something on [How do I update ROS?](https://answers.ros.org/question/10419/how-do-i-update-ros/), but I haven't tried it yet.Is this tutorial right?

pfankhauser, 2017-10-11:
Ok great, looks good.

closed 2017-10-11 by pfankhauser

hujun1413, 2017-10-11:
wait for a moment, I got a new problem.

I clone the older release 1.5.2，and it works on Indigo 1.11.20 and Indigo 1.11.21!

**Then I tried the newest grid_map version on Indigo 1.11.21, and I got new error**:
```
In file included from /opt/ros/indigo/include/filters/filter_chain.h:35:0,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/include/grid_map_demos/FiltersDemo.hpp:14,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/src/filters_demo_node.cpp:9:
/opt/ros/indigo/include/pluginlib/class_loader.h:51:21: error: expected type-specifier
   using UniquePtr = class_loader::ClassLoader::UniquePtr<T>;
                     ^
In file included from /opt/ros/indigo/include/filters/filter_chain.h:35:0,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/include/grid_map_demos/FiltersDemo.hpp:14,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/src/filters_demo_node.cpp:9:
/opt/ros/indigo/include/pluginlib/class_loader.h:112:9: error: ‘UniquePtr’ does not name a type
         UniquePtr<T> createUniqueInstance(const std::string& lookup_name);
         ^
In file included from /opt/ros/indigo/include/pluginlib/class_loader.h:323:0,
                 from /opt/ros/indigo/include/filters/filter_chain.h:35,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/include/grid_map_demos/FiltersDemo.hpp:14,
                 from /home/cutcat/catkin_ws_hujun/src/grid_map/grid_map_demos/src/filters_demo_node.cpp:9:
/opt/ros/indigo/include/pluginlib/class_loader_imp.h:157:3: error: ‘UniquePtr’ does not name a type
   UniquePtr<T> ClassLoader<T>::createUniqueInstance(const std::string& lookup_name)
   ^
```
It seems like ros source code `class_loader.h` have some problem.