ANYbotics/grid_map #136: Is GridMap rviz rendering one cell short ?

artivis, 2017-10-15:
Not sure if it is actually an issue but when overlaying an occupancy grid and a grid map, the two plots mismatch - see attached screenshot. Visually it looks like grid map is one cell short.  
I noticed that in the [GridMapVisual](https://github.com/ethz-asl/grid_map/blob/master/grid_map_rviz_plugin/src/GridMapVisual.cpp#L132) class cells are iterated up to `rows\cols` **-1**.  

Is this the intended behavior ? What is the rational behind this ?

[image]

Simple 50x50 square fully marked as occupied. The occupancy grid overflow on each side (black border). Moreover, the grid map displayed is only 49x49.

[image]

pfankhauser, 2017-10-15:
Occupancy grid renders a square per grid map entry and the grid map rviz visualization creates a mesh where the vertices are located in the cell center. That's why you see a half a cell width difference. Does this answer your question?

artivis, 2017-10-15:
Yes it does. Thanks !
Closing this.

closed 2017-10-15 by artivis