ANYbotics/grid_map #141: Regression with OpenCV 3.3 in Kinetic

tfoote, 2017-11-15:
OpenCV 3.3 was released and there's a regression here: @vrabaud do you know the fix for this off hand?

Re: https://github.com/ros-infrastructure/rep/pull/140

http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__grid_map_demos__ubuntu_xenial_amd64__binary/
```
lboost_atomic -lpthread -lconsole_bridge -Wl,-rpath,/opt/ros/kinetic/lib:/usr/lib/x86_64-linux-gnu/hdf5/serial/lib: 
00:07:32.022 CMakeFiles/opencv_demo.dir/src/opencv_demo_node.cpp.o: In function `main':
00:07:32.023 /tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/src/opencv_demo_node.cpp:57: undefined reference to `cv::namedWindow(cv::String const&, int)'
00:07:32.023 /tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/src/opencv_demo_node.cpp:73: undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
00:07:32.025 /tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/src/opencv_demo_node.cpp:74: undefined reference to `cv::waitKey(int)'
00:07:32.026 collect2: error: ld returned 1 exit status
00:07:32.029 CMakeFiles/opencv_demo.dir/build.make:334: recipe for target 'devel/lib/grid_map_demos/opencv_demo' failed
00:07:32.029 make[4]: *** [devel/lib/grid_map_demos/opencv_demo] Error 1
00:07:32.029 make[4]: Leaving directory '/tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/obj-x86_64-linux-gnu'
00:07:32.030 CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/opencv_demo.dir/all' failed
00:07:32.030 make[3]: *** [CMakeFiles/opencv_demo.dir/all] Error 2
00:07:32.030 make[3]: Leaving directory '/tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/obj-x86_64-linux-gnu'
00:07:32.030 Makefile:130: recipe for target 'all' failed
00:07:32.030 make[2]: *** [all] Error 2
00:07:32.031 make[2]: Leaving directory '/tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2/obj-x86_64-linux-gnu'
00:07:32.031 dh_auto_build: make -j1 returned exit code 2
00:07:32.031 	cd /tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2
00:07:32.033 debian/rules:36: recipe for target 'override_dh_auto_build' failed
00:07:32.033 make[1]: *** [override_dh_auto_build] Error 2
00:07:32.033 make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-grid-map-demos-1.5.2'
00:07:32.037 debian/rules:23: recipe for target 'build' failed
00:07:32.037 make: *** [build] Error 2
00:07:32.038 dpkg-buildpackage: error: debian/rules build gave error exit status 2
00:07:32.041 E: Building failed
```

tfoote, 2017-11-15:
@mikaelarguedas fyi

miquelmassot, 2017-11-15:
To me it looks as if it lacked opencv highgui library linking. Adding OpenCV linking [here](https://github.com/ethz-asl/grid_map/blob/master/grid_map_demos/CMakeLists.txt#L149) might solve it.

pierrefranklin, 2017-11-15:
Yes, that's exactly what's happening. I think when cv_bridge switched to using specific components from OpenCV instead of the entire OpenCV, grid_map lost those libraries. 

I've already made a pull request for this https://github.com/ethz-asl/grid_map/pull/140. However there is a different issue with ros Times in grid_map_ros which is causing the tests to fail, I'm currently working on solving that.

pierrefranklin, 2017-11-16:
The issues have been solved in pull request https://github.com/ethz-asl/grid_map/pull/140. This should get merged in pretty soon.

pfankhauser, 2017-11-16:
Done 👍

closed 2017-11-16 by pfankhauser

mikaelarguedas, 2017-11-16:
Nice! thanks @pierrefranklin @pfankhauser !
This will require a new release for the binaries to be part of the next sync