ANYbotics/grid_map #142: ERROR: cannot launch node of type [grid_map_demos/simple_demo]: can't locate node [simple_demo] in package [grid_map_demos]

Mosquito19, 2017-11-15:
I build from the source code successly,but when I run : 
$roslaunch grid_map_demos simple_demo.launch
It shows:
ERROR: cannot launch node of type [grid_map_demos/simple_demo]: can't locate node [simple_demo] in package [grid_map_demos]
ERROR: cannot launch node of type [grid_map_visualization/grid_map_visualization]: can't locate node [grid_map_visualization] in package [grid_map_visualization]
process[rviz-4]: started with pid [27004]
[ERROR] [1510757591.418195526]: PluginlibFactory: The plugin for class 'grid_map_rviz_plugin/GridMap' failed to load.  Error: Could not find library corresponding to plugin grid_map_rviz_plugin/GridMap. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

pierrefranklin, 2017-11-24:
Did you solve this?

Just form the errors, I would guess you need to resource the setup.bash file in our devel folder after building for the first time.

zhengzh, 2018-06-20:
```bash
catkin_make --only-pkg-with-deps grid_map_demos
catkin_make --only-pkg-with-deps grid_map_visualization
catkin_make --only-pkg-with-deps grid_map_rviz_plugin
```

immuhammadadil, 2019-07-20:
You just need to change the path of the dataset