ANYbotics/grid_map #151: Incremental publishing of high resolution grid map

laxnpander, 2017-12-04:
Hey there,

thanks for your nice package, awesome work! I am looking for a way to handle large, high resolution maps (e.g. 10000x10000 Elements). In default I guess, I need to allocate the full final size at once and then publish the whole map with every update. However this sounds very ineffective to me, in case I have a robot that has very limited view of the total map. 
My question therefore:
- Is GridMap the right package for such a task? Or is it only designed for dynamic, local mapping?
- I have a modification in mind, that uses multiple grid maps of fixed size, and updates only the one that has undergone change. So the 10000x10000 map might be split up to 100 1000x1000 grid maps. Smart? Stupid? Do you think the overhead would be too much?

Thanks for your opinion and best regards,
Alex

artivis, 2017-12-05:
The `GridMap` class class offers the possibility to retrieve a sub-map, you could thus retrieve and publish a sub-map centered on your robot  : 

```cpp
grid_map::GridMap grid_map;
...
grid_map::Position robot_pose(0., 0); // The current position of your robot
grid_map::Length sub_map_length(4, 4); // The sub-map dimension in m.
bool success = false; // Whether the submap was retrieved successfully 
grid_map::GridMap sub_map = grid_map.getSubmap(robot_pose, sub_map_length, success);

if (success)
{
  // Do the publishing
}
```

laxnpander, 2017-12-05:
I see, thanks! I am not so familiar with RVIZ or 3D visualization at all, what does constant publishing of submaps do to the performance? Will it try to render let's say 20 submaps even in overlapping areas, or will it treat the grid map as one total object?

closed 2017-12-20 by laxnpander