ANYbotics/grid_map #154: Seg Fault on 2 sides of 4 boundary sides

alicegong, 2017-12-06:
I found that when I have the origin and world length specified, if I try to access a coordinate on the boundary of the world, it’d seg fault. E.g., origin = (0,0), world length = (width = 50, height = 20), resolution = 0.05, then I cannot access (any x, -10), or (-25, any y) but (any x, 10) and (25, any y) were fine. It took some digging into your source code for me to understand how the numerical coordinate values actually corresponded to individual grids, so it made sense and I don’t think it was necessarily a bug in your code (which is very well written and organized, by the way), but rather that you just never took that nuance into consideration?

pfankhauser, 2017-12-08:
Hi Alice,
Thanks for reporting this behavior. You are right, I have not taken this special case into consideration since we’ve never run into this issue. I you have a suggestion for a change in code, I would be happy to take a look at it in a pull request.
Best, Péter

alicegong, 2017-12-10:
Sure. I'll fork it and modify, and let you know when I've made a pull request!