ANYbotics/grid_map #155: [grid_map_cv] opencv_photo-NOTFOUND

stonier, 2018-01-02:
Off on a wild goose chase trying to track down this linking problem for `libgridmap_cv.so`, so throwing some information here to save the next person some time till the problem is fixed upstream.

i.e. probably nothing needed to fix here, just close when OpenCV3.4 has been build for ROS Kinetic. Discussion and updates should be found in [this thread](https://github.com/ros-perception/vision_opencv/issues/193).

artivis, 2018-01-08:
Wish I had found this post half an hour ago...
Anyway, on my machine (Ubuntu 14.04 + Indigo) both OpenCV2.4 and ros-indigo-opencv3 are installed.
It seems that the `find_package(cv_bridge)` pulls OpenCV2 (make sense) while `find_package(OpenCV REQUIRED COMPONENTS opencv_photo)` pulls OpenCV3 (make sense ??!).  
Leading (much) later on and a couple package on top to a sweet segfault...

I could comment the `find_package(OpenCV...` as a temporary fix.

ThomasEmter, 2018-04-05:
Hi @stonier. Did you try cleaning your workspace as mentioned in the thread by @Triocrossing? This solved the issue of `No rule to make target '/opt/ros/kinetic/lib/libopencv_photo3.so.3.3.1'`. Perhaps this also helps in your case?

Hi @artivis. I think your issue has another reason: the parallel usage of OpenCV3 in ROS Indigo. It is possible, but the usage of `cv_bridge` segfaults. See for example [here](https://github.com/ros-perception/vision_opencv/issues/91).