ANYbotics/grid_map #157: Performance: getPosition(Index) is super slow

laxnpander, 2018-01-03:
Hey hey,

I currently have an iteration of about 10 000 elements in a submap which runs in less than 0.1 seconds. However, as soon as I add "getPosition(*iterator, position)" into the loop, the performance drops to 0.5 seconds and worse. This is not really acceptable for me, especially as it should be quite an easy computation? Whats going on there?

Thanks for your help and best regards,
Alex

taogashi, 2018-08-14:
The same for `getIndex(position, index)`. This is the Gperftools result:
[image]

brunoeducsantos, 2022-03-15:
Hi @maximilianwulf  ,
I have the same performance issue. 
Iterating over a grid with 10 000 cells and getting the position inside the iterator results in a 40 ms. 
```
   for (grid_map::GridMapIterator iterator(grid_map); !iterator.isPastEnd();
         ++iterator) {
      const int i = iterator.getLinearIndex();
      float res = data(i);
      grid_map::Position currentCellPosition;
      grid_map.getPosition(*iterator, currentCellPosition);
}
```

Is there a work around to improve this performance?
Cheers,
Bruno