ANYbotics/grid_map #158: PointCloud2 to Gridmap?

aybakana, 2018-01-18:
Can I just convert a PointCloud2 message into Gridmap? How can I do that?

pfankhauser, 2018-01-18:
Hi @aybakana! A point cloud and a grid map are very different formats, which cannot just be converted. You’ll have to do a mapping from the individual points of the point cloud to the values of the cells in the grid map. We have an [elevation mapping](https://github.com/ethz-asl/elevation_mapping) framework that could help you with this.

aybakana, 2018-01-18:
I converted PointCloud2 message into Octomap message using _octomap_server_ node and then Octomap message into Gridmap message using _gridmap_to_octomap_demo_ node. It seems okay. This was what I meant. Can you not add a node to do this conversion at once? I am still confused what you meant by grid map and point cloud are very different formats. Can you give me a source which will help me understand better?

JesusCoyotzi, 2019-01-07:
I realize this is an old issue but i think it could be a nice feature to be able to load a cloud and turn it into an occupancy grid.
In the meantime and if anyone needs to do this you can use octomap_mapping to turn a pointcloud into and octomap and then project it into a conventional 2D occupancy grid.

sherifsas, 2019-02-27:
@aybakana could you tell me how to convert octomap message into gridmap using gridmap_to_octomap_demo ?

aybakana, 2019-02-28:
Sorry, I've mistakenly written gridmap_to_octomap, it must be [octomap_to_gridmap_demo_node](https://github.com/ANYbotics/grid_map/blob/master/grid_map_demos/src/octomap_to_gridmap_demo_node.cpp)

sherifsas, 2019-02-28:
@aybakana Thanks, yeah I am using octomap_to_gridmap_demo_node to convert the octomap published by the octomap_server node. However, I could not send the node output to the octomap_to_grid node. Here is my launch file.

<launch>
  <rosparam command="load" file="$(find grid_map_demos)/config/octomap_to_gridmap_demo.yaml" />

  
  <node pkg="octomap_server" type="octomap_server_node" name="octomap_server" output="screen"/>

  
  <node pkg="grid_map_demos" type="octomap_to_gridmap_demo" name="octomap_to_gridmap_demo" output="screen" />

  
  <node pkg="grid_map_visualization" type="grid_map_visualization" name="grid_map_visualization" output="screen" />

</launch>

sherifsas, 2019-02-28:
[octomap_to_gridmap.txt](https://github.com/ANYbotics/grid_map/files/2914842/octomap_to_gridmap.txt)

aybakana, 2019-02-28:
You must have forgotten to remap the point cloud input of octomap server. Can you try [this](https://github.com/aybakana/grid_map/blob/master/grid_map_demos/launch/pointcloud_to_gridmap.launch)? I have assumed that your sensor range is 100m and point cloud topic is /velodyne_points and resolution for octomap is 0.05. You can change them as you wish.

LottieWang, 2019-07-22:
I saw there is a directory called **grid_map_pcl**. Is it designed for converting PointCloud to GridMap? If it is, could demo be provided?

KenaHemnani, 2021-03-25:
Hii , I want generate 2D Obstacle grid map via. RGBD image sequentially taken by robot via. first converting each RGBD image into point clouds then projecting point clouds to get local map and then to get entire map of the environment?? Is there such Package available in ROS or PCL Libraries??

maximilianwulf, 2021-05-31:
Yes, https://github.com/ANYbotics/elevation_mapping should do exactly what you want.

flabrosse, 2025-01-17:
Sorry to be reviving this old discussion.  The conversion from ROS PointCloud2 (PC2) can also be done via PCL: ROS PC2 -> PCL PC2 -> PCL PC -> GM (using GridMapPCLLoader).

I have not looked at how efficient this is, but I can't help thinking this is not.  And given that the algorithm to go from a point cloud to a gridmap exists, can't a GridMapROSLoader class be written?  Just a thought.

james-yoo, 2025-01-17:
Check this:
https://github.com/jkk-research/pointcloud_to_grid

ZhongmouLi, 2025-02-19:
I am facing the same issue in ros2 humble. Is there any "direct" solution in ros2?

flabrosse, 2025-02-19:
I'm about to make public a package that does it (amongst other things).  I'll put a link here when it is more ready for public consumption.

flabrosse, 2025-02-19:
I have just made this public: [gradient_cost_plugin](https://github.com/Aber-FAR/gradient_cost_plugin).  If you look at updateBounds() in gradient_cost_layer.cpp you'll see how I convert the pointcloud2 `cloud_transf_` into the GridMap `grid_map_`.

ZhongmouLi, 2025-02-27:
I have found a solution that is to use [octomap_server2](https://github.com/iKrishneel/octomap_server2) to do this.

closed 2025-03-06 by aybakana