ANYbotics/grid_map #159: catkin_make failed in Ubuntu 16.04 (Invoking "make -j8 -l8" failed)

connichiha, 2018-01-26:
I installed the grid_map and its dependencies.
Firstly, I got an error about no package.  I installed the ros_navigation package and fixed it.
```
-- ==> add_subdirectory(grid_map/grid_map_costmap_2d)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "costmap_2d" with
  any of the following names:

    costmap_2dConfig.cmake
    costmap_2d-config.cmake

  Add the installation prefix of "costmap_2d" to CMAKE_PREFIX_PATH or set
  "costmap_2d_DIR" to a directory containing one of the above files.  If
  "costmap_2d" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  grid_map/grid_map_costmap_2d/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'costmap_2d'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "costmap_2d" with
  any of the following names:

    costmap_2dConfig.cmake
    costmap_2d-config.cmake

  Add the installation prefix of "costmap_2d" to CMAKE_PREFIX_PATH or set
  "costmap_2d_DIR" to a directory containing one of the above files.  If
  "costmap_2d" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  grid_map/grid_map_costmap_2d/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
```
------------------------------------------------------------------
Now, I could not fix this problem caused by map_loader( not sure).
```

Base path: /home/vb/catkin_ws
Source space: /home/vb/catkin_ws/src
Build space: /home/vb/catkin_ws/build
Devel space: /home/vb/catkin_ws/devel
Install space: /home/vb/catkin_ws/install
####
#### Running command: "cmake /home/vb/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/vb/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/vb/catkin_ws/install -G Unix Makefiles" in "/home/vb/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/vb/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/vb/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/vb/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/vb/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 15 packages in topological order:
-- ~~  - grid_map (metapackage)
-- ~~  - grid_map_core
-- ~~  - grid_map_octomap
-- ~~  - beginner_tutorials
-- ~~  - grid_map_msgs
-- ~~  - grid_map_cv
-- ~~  - grid_map_ros
-- ~~  - grid_map_filters
[... 487 lines cut ...]
FiltersDemo.cpp:(.text._ZN7filters11FilterChainIN8grid_map7GridMapEEC2ESs[_ZN7filters11FilterChainIN8grid_map7GridMapEEC5ESs]+0x50f): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/filters_demo.dir/src/FiltersDemo.cpp.o: In function `pluginlib::ClassLoader<filters::FilterBase<grid_map::GridMap> >::refreshDeclaredClasses()':
FiltersDemo.cpp:(.text._ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv[_ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv]+0xdb): undefined reference to `class_loader::MultiLibraryClassLoader::getRegisteredLibraries()'
FiltersDemo.cpp:(.text._ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv[_ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv]+0x2cc): undefined reference to `ros::package::getPlugins(std::string const&, std::string const&, std::vector<std::string, std::allocator<std::string> >&, bool)'
FiltersDemo.cpp:(.text._ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv[_ZN9pluginlib11ClassLoaderIN7filters10FilterBaseIN8grid_map7GridMapEEEE22refreshDeclaredClassesEv]+0x51c): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::checkField(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::string const&, unsigned int, unsigned int, bool) const'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::readField(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::string const&, bool, std::string&) const'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::MessageInstance::getMD5Sum() const'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::toHeaderString(ros::Time const*) const'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::appendHeaderToBuffer(rosbag::Buffer&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::writeHeader(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `cv_bridge::getCvType(std::string const&)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `cv_bridge::toCvCopy(sensor_msgs::Image_<std::allocator<void> > const&, std::string const&)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::TopicQuery::TopicQuery(std::string const&)'
/home/vb/catkin_ws/devel/lib/libgrid_map_ros.so: undefined reference to `rosbag::Bag::open(std::string const&, unsigned int)'
collect2: error: ld returned 1 exit status
grid_map/grid_map_demos/CMakeFiles/filters_demo.dir/build.make:197: recipe for target '/home/vb/catkin_ws/devel/lib/grid_map_demos/filters_demo' failed
make[2]: *** [/home/vb/catkin_ws/devel/lib/grid_map_demos/filters_demo] Error 1
CMakeFiles/Makefile2:6614: recipe for target 'grid_map/grid_map_demos/CMakeFiles/filters_demo.dir/all' failed
make[1]: *** [grid_map/grid_map_demos/CMakeFiles/filters_demo.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

connichiha, 2018-01-27:
Hi,everybody. I found the problem and fixed.
Here I share the solutions.
Problem was the version of gcc in my ubuntu.
change the version from 4.9 to 5.0. It worked

closed 2018-01-27 by connichiha

thecreator-hr, 2019-12-10:
I am still not able to resolve the problem after updating gcc
I am getting this

catkin_make
Base path: /home/harsh/catkin_ws
Source space: /home/harsh/catkin_ws/src
Build space: /home/harsh/catkin_ws/build
Devel space: /home/harsh/catkin_ws/devel
Install space: /home/harsh/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/harsh/catkin_ws/build"
####
/usr/bin/cmake -H/home/harsh/catkin_ws/src -B/home/harsh/catkin_ws/build --check-build-system CMakeFiles/Makefile.cmake 0
####
#### Running command: "make -j8 -l8" in "/home/harsh/catkin_ws/build"
####
/usr/bin/cmake -H/home/harsh/catkin_ws/src -B/home/harsh/catkin_ws/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/harsh/catkin_ws/build/CMakeFiles /home/harsh/catkin_ws/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/harsh/catkin_ws/build'
make -f ros_essentials_cpp/CMakeFiles/image_pub_sub_cpp.dir/build.make ros_essentials_cpp/CMakeFiles/image_pub_sub_cpp.dir/depend
make -f ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/build.make ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/depend
make -f ros_essentials_cpp/CMakeFiles/read_video_cpp.dir/build.make ros_essentials_cpp/CMakeFiles/read_video_cpp.dir/depend
make -f ros_essentials_cpp/CMakeFiles/utility_lib.dir/build.make ros_essentials_cpp/CMakeFiles/utility_lib.dir/depend
make -f ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_py.dir/build.make ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_py.dir/depend
make -f ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_eus.dir/build.make ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_eus.dir/depend
make -f ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_lisp.dir/build.make ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_lisp.dir/depend
make -f ros_basics_tutorial/CMakeFiles/rosgraph_msgs_generate_messages_cpp.dir/build.make ros_basics_tutorial/CMakeFiles/rosgraph_msgs_generate_messages_cpp.dir/depend
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_basics_tutorial /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_basics_tutorial /home/harsh/catkin_ws/build/ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_py.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_basics_tutorial /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_basics_tutorial /home/harsh/catkin_ws/build/ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_eus.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_essentials_cpp /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_essentials_cpp /home/harsh/catkin_ws/build/ros_essentials_cpp/CMakeFiles/utility_lib.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_basics_tutorial /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_basics_tutorial /home/harsh/catkin_ws/build/ros_basics_tutorial/CMakeFiles/roscpp_generate_messages_lisp.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_basics_tutorial /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_basics_tutorial /home/harsh/catkin_ws/build/ros_basics_tutorial/CMakeFiles/rosgraph_msgs_generate_messages_cpp.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_essentials_cpp /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_essentials_cpp /home/harsh/catkin_ws/build/ros_essentials_cpp/CMakeFiles/read_video_cpp.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_essentials_cpp /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_essentials_cpp /home/harsh/catkin_ws/build/ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/harsh/catkin_ws/build'
cd /home/harsh/catkin_ws/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/harsh/catkin_ws/src /home/harsh/catkin_ws/src/ros_essentials_cpp /home/harsh/catkin_ws/build /home/harsh/catkin_ws/build/ros_essentials_cpp /home/harsh/catkin_ws/build/ros_essentials_cpp/CMakeFiles/image_pub_sub_cpp.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/harsh/catkin_ws/build'
make[2]: Leaving directory '/home/harsh/catkin_ws/build'
[... cut ...]
make[2]: *** [ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/src/topic03_perception/cpp/open_copy.cpp.o] Error 1
make[2]: Leaving directory '/home/harsh/catkin_ws/build'
CMakeFiles/Makefile2:1038: recipe for target 'ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/all' failed
make[1]: *** [ros_essentials_cpp/CMakeFiles/open_copy_cpp.dir/all] Error 2
make[1]: Leaving directory '/home/harsh/catkin_ws/build'
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

maximilianwulf, 2019-12-19:
@thecreator-hr can you try:

`catkin build grid_map`

thecreator-hr, 2019-12-27:
what I did was changed this
`image = imread("/home/riotu/catkin_ws/src/ros_essentials_cpp/src/topic03_perception/images/chess.jpg", CV_LOAD_IMAGE_COLOR); // Read the file "image.jpg".` to 
imread("/home/riotu/catkin_ws/src/ros_essentials_cpp/src/topic03_perception/images/chess.jpg", 1); // Read the file "image.jpg".

maximilianwulf, 2020-01-06:
Thanks for the update, feel free to reopen if there are other issues.