ANYbotics/grid_map #167: Grid Size Rviz Limitations

erichahn, 2018-06-14:
When using an occupancy map of size 300 by 300 meters rviz never slows down. When attempting to convert at 666 x 666 pixel image to a grid map of size 300 by 300 meters at 0.5 meter resolution I get about a message a second, compared to the typical 20 a second possible with occupancy maps.

Any idea what would be contributing to this severe slowdown?

pfankhauser, 2018-06-15:
Hi @erichahn, are you building from source? Can you confirm that you're building in RELEASE mode?

erichahn, 2018-06-18:
I am installing using sudo apt install ros-kinetic-grid-map, I am building in release mode. It appears to be an rviz plugin slowdown as issue only develops when attempting to visualize.

skohlbr, 2018-09-04:
I'm experiencing the same issue. For small grid_maps, the plugin works fine, but it goes as far as completely locking up my machine, requiring a restart (!) for if the map is "too large". This is on 16.04/Kinetic/latest from .debs. Don't have time for a comprehensive test series, but will update this ticket as I come across datapoints :)

ulassbin, 2021-07-19:
I am having a similar problem with a base_map of 2639 x 3500 size. My gridmap has 5 layers, and the topic is published at a  rate of 0.2 hz.  The code processing this gridmap is around %4 in cpu usage, whereas rviz is at %90-100. I am  not sure if this is a gpu issue, I have controlled the resource usage with htop, maybe a more detailed investigation would point out to the cause.

I have noticed there is a parameter called 'Autocompute Intensity Bounds', can this(or something similar) be running at background eventhough it is switched off or another transformer for color is selected? I couldn't find time to check the source code yet.

juanluishortelano, 2023-07-07:
Same issue here. ROS Melodic, Ubuntu 18.04, Gridmap 1.6.4. Pretty slow on rviz.

juanluishortelano, 2023-07-13:
Found out that unticking "Show Grid Lines" improves performance

cedrichld, 2025-09-29:
This is still an issue in ROS2 Jassy and Gazebo Harmonic. I am using a 513x513 8bit grayscale heightmap (I believe 2n+1 is what's best for Gazebo) of 30mx30mx4.5m and see an issue when scaling the map in x/y or if i increase the map resolution. If I try the 1025x1025 version of the heightmap it becomes ridiculously slow (even when I take out the grid lines). Is there a way we could fix this / use a different plugin?