ANYbotics/grid_map #169: Gridmap for large areas

murikill, 2018-06-19:
Hello, 

it is possible to use grid_map for large areas (memory efficient)? 
For example 10000 x 10000 gridcells.

Thank you

pfankhauser, 2018-06-19:
Hi @murikill, Grid Map builds on Eigen dense matrices, so this should scale well for large data. However, you will have to take care to manipulate the data efficiently, for more details on this see [here](https://github.com/ethz-asl/grid_map#iterators). Please let me know if you gain more experience.

bryanibit, 2018-07-06:
@murikill Did you finish the large scale mapping?

murikill, 2018-07-10:
@bryanibit I'm using grid_map for small maps. For large matrices you need to have a lot of memory on your pc.

bryanibit, 2018-07-10:
@murikill can you provide some thoughts about mosaic pointclouds to a large map with grid_map?

closed 2018-07-11 by murikill