ANYbotics/grid_map #170: transform grid_map frames

erichahn, 2018-06-19:
Is it possible to transform grid_map frames from say base_link to odom on a mobile robot platform? I understand how to set a new position, but setting any concept of orientation about the center of the grid appears to not be possible.

When trying to simply do the transform on a grid_map_msg using tfListener the code refuses to compile. Guessing it's not supported as it's a custom type without support.

Thoughts on transforming between frames for either the grid_map or grid_map_msg?

erichahn, 2018-06-20:
I have a pointcloud with frame base_link. When I create an intensity image of the point cloud, and then create a grid_map from the image, I get the tf from base_link to odom. I then apply the transform to the grid_map. The translation works properly, but the orientation is not correct at all.

Secondly, if I then try and create an occupancy grid from the grid_map the position transformation is retained, but not the z height. It defaults to 0.

erichahn, 2018-06-21:
Between forcing the pose orientation to remain fixed at 0, be set to the transform between frames, or manually adjust roll, pitch, and yaw using dynamic_reconfigure and converting back to a quaternion it seems like the pose orientation is not used at all. Is this expected?

marco-tranzatto, 2019-01-08:
Hi @erichahn I need to do a similar thing. What about the following procedure:

1. convert `grid_map` object to a `cv_image`, 
2. apply openCV functions to obtained the desired transformation,
3. convert the result back to `grid_map`?

I guess there can be some artifacts from the procedure, but I see no better way. What do you think @pfankhauser ?

maximilianwulf, 2022-06-15:
Does https://github.com/ANYbotics/grid_map/blob/master/grid_map_core/include/grid_map_core/GridMap.hpp#L324 help you ?

martinakos, 2022-06-15:
I need to transform the gridmap out of plane, as my world frame is set by a VIO system and may not be axis aligned with the floor. I solved the problem by creating a moving tf frame for where I want my gridmap in the world and using that one as frame_id then setting the pose to zero.

maximilianwulf, 2022-06-17:
Thanks, then this issue can be closed?

martinakos, 2022-06-17:
Yes, it can be closed. Thanks.

closed 2022-06-17 by maximilianwulf

martinakos, 2022-06-23:
I have a problem with the separate moving tf frame solution. I first broadcast a moving tf frame for the elevation map. Then I publish the GridMap using the frame_id of my moving frame. Then I publish some other marker information that is synchronised with the elevation map, but that I can publish on the world frame. In Rviz I always see the elevation map with a delay with respect to the other marker information despite both being published at the same time, with same timestamp. From what I could find out the tf system always has a bit of delay. " The delay comes from the fact that tf needs to receive transformations from different broadcasters in the system." https://wiki.ros.org/tf/Troubleshooting#Checking_the_Timestamp 
So this can explain why I always see the elevation map behind the corresponding markers.  

I guess this is a feature request, but would it be possible to make the Rviz GridMap plug-in to use the GridMap message Pose so that I could use the world frame? Then I wouldn't have a delay between elevation map and the other related markers.

martinakos, 2022-08-05:
Shall I repost my last message as a new feature request issue?