ANYbotics/grid_map #171: ROS Melodic release?

bochen87, 2018-06-22:
Hi,

Are you planning on releasing versions for melodic?

pfankhauser, 2018-07-26:
Hi @bochen87, 
Yes, Grid Map will also be released for the latest versions of ROS, but I cannot give you a date right now. In the meanwhile, have you tried to build Grid Map from source?
-Peter

flbeck, 2018-07-30:
Hi,

I recently tried to use grid_map with ROS Melodic on Ubuntu bionic. The only issue I faced, when compiling from source, was that the RViz plugin did not work because of the migration to tf2 (see [rviz migration](http://wiki.ros.org/melodic/Migration)). Simply replacing the `tf` namespace with `tf2_ros` was sufficient for it to work. However, I don't think it is backward compatible this way.

I can do a pull request if you are interested.

pfankhauser, 2018-07-30:
Thanks for checking this out, since `tf2` also exists on older versions of ROS, this might actually work. Yes please, go ahead and do a PR.

redkite, 2018-12-04:
@pfankhauser Are you still looking into this? The PR has been open for quite a while.

I could also supply a PR with an updated `frame_manager.h` from melodic which includes deprecation warnings from melodic to not use boost smart pointers anymore.

In any case a push to a melodic release would be highly appreciated.

MirkoFerrati, 2018-12-06:
I am also looking forward for the melodic update! I went forward with #179  and fixed the info/header issues so that frame_manager.h is no longer needed since it was just a matter of replacing info.header with header and changing the messages. @pfankhauser do you think it makes sense to comply with rviz use of message headers? do you want the PR?

samuelba, 2019-02-27:
I did the release for melodic. Should be in the next sync. https://github.com/ros/rosdistro/pull/20399

closed 2019-02-27 by samuelba