ANYbotics/grid_map #175: Rviz extremely slow when showing grid maps

ndalmedico, 2018-07-09:
Hi, I don't know if I am missing a depency or something, but the grid map displays slows down Rviz to 1fps, even when the map is small. Octomaps on the other hand are ploted in 30fps like always.

Using the simple demo as example. I can see the octomap generated from the grid map in 30fps, but enabling the grid map message immediatly brings the fps to 1. Rostopic hz on the grid map message gives me ~20hz. Also, the 'surface normals' and 'elevation cells' topics are not published. My guess is I am missing something, but the dependencies looks right (rosdep says it's ok), make does not give any errors or warning. I tried on a clean Ubuntu 14.04 with ROS Indigo to be sure and the same thing happened. 

Were there changes to grid map rviz plugin? Am I missing a package? Or is this how it was supposed to be?

fmascarich, 2018-07-23:
When you're building the package, make sure you use -DCMAKE_BUILD_TYPE=Release. Mine was updating at around 0.25 hz before I remembered to set the cmake flag.

ndalmedico, 2018-07-24:
YES! THANK YOU VERY MUCH! That was it! From <1 to 30 fps

Closing the issue

closed 2018-07-24 by ndalmedico