ANYbotics/grid_map #178: Any plan for ROS2?

Myzhar, 2018-07-24:
Hi,
what is your plan about a ROS2 version of grid_map?

I'm adding grid_map support in a sensor node with an eye on ROS2 and knowing what you are going to do with ROS2 is really important for my decisions.

Thank you
Walter

pfankhauser, 2018-07-24:
Thanks for reaching out @Myzhar. We are planning to release a version of Grid Map for ROS 2, but I cannot give you a release date. If you have some spare time to look into the changes that will be required, it would help us to speed up the process.

Myzhar, 2018-07-24:
Since GridMap library is ROS indipendent I think that the only required changes are relative to the ROS nodes that you developed. However I will let you know.

pfankhauser, 2018-07-24:
Ok great, I'll keep this issue open for the discussion.

bpwilcox, 2018-08-06:
Hi, I'm with a team working on the new [ROS2 Navigation package](https://github.com/ros-planning/navigation2). We are looking to support grid_map among other map representations (see issue: https://github.com/ros-planning/navigation2/issues/18)  in the new navigation stack, so we would be greatly interested in hearing any progress made on an official ROS2 version. As @Myzhar  mentioned, there may not be much ROS dependency, but when we cross that bridge, we'll be open to share what we've learned.

bpwilcox, 2019-09-30:
I'd like to revive this thread. As we are discussing the current design of core interfaces in the navigation2 stack (see https://github.com/ros-planning/navigation2/issues/1173) , we should consider the support of map representations outside of costmap_2d. 

@pfankhauser Has there been any investigation into this package for ROS2?

closed 2021-05-17 by SteveMacenski