ANYbotics/grid_map #181: Bandwidth tuning

mescarra, 2018-08-17:
So, the domain I'm currently working at requires huge maps with high resolutions. The problem is that I'm connecting to my robot through a wireless network. I also have to stream video, but currently the main bottleneck in my project is the grid_map. So far, I've reduced publishing frequency and map resolution, but still I'm way off the requirements.

Is there any way to compress grid_map topics (even in detriment to performance)?

skohlbr, 2018-08-17:
There are some external tools you could use, for instance
 * https://github.com/Eurecat/ros_topic_compressor
*  https://github.com/tu-darmstadt-ros-pkg/topic_proxy
* https://github.com/AIS-Bonn/nimbro_network.

They all allow compression using BZ2 or LZ4, which can compress grid map data with often a fairly regular structure (lots of NaNs etc.) very well.

pfankhauser, 2018-08-17:
Another idea would be to convert the grid maps to an image and use the ROS [image_transport](http://wiki.ros.org/image_transport) for compressed streaming, but I haven't tried this myself.

Please, let us know about your experience for the things you try out.

mescarra, 2018-08-21:
Hello guys,

I tried the **ros_topic_compressor** and it works perfectly. Easy to set up and achieves a very high compression rate on the grid_map topic.

Thank you very much for the advice!

closed 2018-08-21 by mescarra