ANYbotics/grid_map #186: getPosition(Index, Position) returns wrong results if the map has been moved

ChristofDubs, 2018-09-25:
Hello there,

I am a generally happy user of the grid_map library and think it offers a very sensible set of functionalities.

However, I came across an issue today which I traced down to be an error in the function `bool GridMap::getPosition(const Index& index, Position& position) const`. I started writing a unit test in order to reproduce the error with a few lines of code: 

```
#include <gtest/gtest.h>

#include <grid_map_core/grid_map_core.hpp>
#include <iostream>

#define TOL 1e-3

TEST(Gridmap, getPositionFromIndex) {
    grid_map::GridMap map({"layer1", "layer2"});

    const grid_map::Length gridMapSize(20, 20);
    const double resolution = 0.2;

    map.setGeometry(gridMapSize, resolution);

    const grid_map::Position gridMapCenter(4.2, 4.6);

    map.move(gridMapCenter);

    grid_map::Position pos = map.getPosition();
    EXPECT_NEAR(pos(0), gridMapCenter(0), TOL);   // passes
    EXPECT_NEAR(pos(1), gridMapCenter(1), TOL);   // passes

    const grid_map::Index index1(36,76), index2(36,77);

    if(map.getPosition(index1, pos))

        std::cout << pos(0) << " " << pos(1) << std::endl;    // outputs "2.7 -5.3"

    if(map.getPosition(index2, pos))

        std::cout << pos(0) << " " << pos(1) << std::endl;    // outputs "2.7 14.5"

}
```

The reported position for the first point is `[2.7, -5.3]`, and for the second one `[2.7, 14.5]`, which is clearly wrong since those are neighboring points. I traced the issue further down to the function `getPositionFromIndex()` in `grid_map_core/src/GridMapMath.cpp`, in which the part `getIndexVectorFromIndex(index, bufferSize, bufferStartIndex)(1)` equals to `-99` for `index1` and `0` for `index2`.

It would be nice if you could quickly look into this. 

Thank you.

skohlbr, 2018-10-02:
Another issue that apparently only manifests on a moving map is described in #180. 

I also ran into the issue you describe and discovered that calling
`grid_map::wrapIndexToRange(index, grid_map_.getSize());` fixes these kinds of crashes. I basically apply this function to any index I use per default when operating on moving map and don't see this kind of crash anymore. Not quite sure if that's a hack or the recommended approach. It certainly would be beneficial to document that it is required (or what other options there are).

remod, 2020-07-02:
I started using grid map some days ago, allow me to share my insights.

I assume that the behavior you observe is correct. For performance reasons, moving a grid map does not move the underlying Eigen matrix, but only the mapping from position to index (through modifying the starting index). Therefore, a specific index (x,y) will not point to the same position anymore after moving the grid map. This means that two "neighboring" indices (x,y) and (x,y+1) do not necessarily need to point to "neighboring" positions anymore, as they might "wrap around" as in your example.

Therefore I think that one should not do any assumptions on where indices point to, but always derive the by calling `getIndex(..)` giving the desired position. Once the map is moved again, all previous indices will point to new positions.

maximilianwulf, 2020-07-03:
To additionally visualize the behavior, I `move()` the grid maps before iterating in the `grid_map_demos` package.

In the first case, you see the values that you would get by getting the `normal` index: 

[image]

The `move()` operation changes the order of indexing. That is because of the underlying two-dimensional circular buffer. Using this index is a bit inconvenient, but `grid_map` provides different functions to access grid maps independent of the underlying circular buffer. 

For instance in the `GridMapIterator` demo we can replace `iterator.getLinearIndex()` with `iterator.getUnwrappedIndex()`. This will then procude the following result: 

[image]

So depending on the use case you want to do the one or other. I agree that it is not obvious in the documentation, we will update the documentation accordingly. But `getPosition()` should work as intended.

closed 2020-07-03 by maximilianwulf