ANYbotics/grid_map #192: addLayerFromImage height scaling and map duplicate

BrunoEduardoCSantos, 2018-11-28:
Hi @pierrefranklin ,
My goal is converting image to grid map . For this purpose I used `addLayerFromImage` as follows:
```

cv::Mat image= cv::imread(file, CV_LOAD_IMAGE_COLOR);
    bool res=  grid_map::GridMapCvConverter::addLayerFromImage<unsigned short, 3>(image,"elevation",gridMap);
```

Although, my map shows duplicate and having height scaling. 
[image]

Is it a problem with height scaling conversion?
Regards,
Bruno

joshjowen, 2018-12-13:
It appears to me that you're file is saved as 16-bit but you are loading it with cv::imread as 8-bit. try changing
` cv::imread(file, CV_LOAD_IMAGE_COLOR);` 
to 
`cv::imread(file, CV_LOAD_IMAGE_COLOR | CV_LOAD_IMAGE_ANYDEPTH);`

BrunoEduardoCSantos, 2018-12-13:
It worked out @joshjowen ! Thanks a lot!

closed 2018-12-13 by BrunoEduardoCSantos