ANYbotics/grid_map #209: image pointer never deleted in SDF

BeatScherrer, 2019-04-11:
The image pointer created at https://github.com/ANYbotics/grid_map/blob/master/grid_map_sdf/src/SignedDistanceField.cpp#L84 is never deleted.

PS: Thanks for your effort put in this great library

BeatScherrer, 2019-04-11:
Further I'm not quite sure why the SDF encorporates a vector of matrices, the dt transforms binary images to distance fields. I don't manage to see the purpose of this additional dimensionality since the dt is only performed in 2D. The amount of layers added is proportional to the indices in the input matrix.

Unfortunately there is no documentation for this module.

maximilianwulf, 2019-04-17:
Hi @BeatScherrer, did run into a segmentation fault? 

Can you propose a  pull request that fixes the issue?