ANYbotics/grid_map #213: Faster polygon iterator?

snieh, 2019-04-23:
Is there a way to run the polygon iterator faster?
My grid_map has a size of 200x200 with resolution of 0.25, which is quite big (640.000 cells) and a even bigger polygon which intersects the gird_map. Around about 50.000 cells inside the polygon of the grid_map (globalMap) then need to be set to an value, which requires often up to 3 seconds...
 
for (grid_map::PolygonIterator iterator(globalMap, polygon); !iterator.isPastEnd(); ++iterator) {
        globalMap.at("test", *iterator) = 0.0; 
    }

//Also tried this way, which needs same computation time:
 auto& data_to = globalMap["test"];
 for (grid_map::PolygonIterator iterator(globalMap, polygon); !iterator.isPastEnd(); ++iterator) {
    float& value_to = data_to( (*iterator)[0] , (*iterator)[1] );
    value_to = 0.0;
  }

Any suggestions? Thanks in advance.

zoojing, 2019-06-08:
The provided iterator would call function is_Inside for each cell to judge if it is out of range. however, is_inside has O(n^2)  complexity, which means the polygon iterator has O(n^3) complexity.
In my opinion, diy polygon scan or use bfs would be a nice idea. 
Ps: the index return from get_index is not a normal sequence when you move the grid map.
Good Luck!

AlexReimann, 2022-03-28:
Maybe interesting for others:

https://github.com/tu-darmstadt-ros-pkg/grid_map_proc/pull/1

maxime-clem, 2022-05-30:
I reimplemented the `PolygonIterator` using a scan line algorithm like @zoojing suggested.
It performs several orders of magnitude faster and behaves almost exactly like the current implementation. Differences only occur when a polygon edge exactly crosses the center position of a grid cell.
https://github.com/autowarefoundation/autoware.universe/tree/main/common/grid_map_utils