ANYbotics/grid_map #214: sudo apt-get install ros-$ROS_DISTRO-grid-map

gouyiyong, 2019-06-14:
run sudo apt-get install ros-$ROS_DISTRO-grid-map, the problem is here, how can i fix it? thanks 
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-core amd64 1.6.1-0xenial-20190227-080735-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:2 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-cv amd64 1.6.1-0xenial-20190320-152756-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:3 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-msgs amd64 1.6.1-0xenial-20190320-130200-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:4 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-ros amd64 1.6.1-0xenial-20190320-161329-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:5 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-filters amd64 1.6.1-0xenial-20190320-171012-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:6 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-loader amd64 1.6.1-0xenial-20190320-162422-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:7 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-octomap amd64 1.6.1-0xenial-20190227-095814-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:8 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-rviz-plugin amd64 1.6.1-0xenial-20190320-181602-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:9 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-visualization amd64 1.6.1-0xenial-20190320-162423-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:10 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map-demos amd64 1.6.1-0xenial-20190320-183326-0800
  404  Not Found [IP: 64.50.233.100 80]
Err:11 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-grid-map amd64 1.6.1-0xenial-20190320-184931-0800
  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-core/ros-kinetic-grid-map-core_1.6.1-0xenial-20190227-080735-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-cv/ros-kinetic-grid-map-cv_1.6.1-0xenial-20190320-152756-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-msgs/ros-kinetic-grid-map-msgs_1.6.1-0xenial-20190320-130200-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-ros/ros-kinetic-grid-map-ros_1.6.1-0xenial-20190320-161329-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-filters/ros-kinetic-grid-map-filters_1.6.1-0xenial-20190320-171012-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-loader/ros-kinetic-grid-map-loader_1.6.1-0xenial-20190320-162422-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-octomap/ros-kinetic-grid-map-octomap_1.6.1-0xenial-20190227-095814-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-rviz-plugin/ros-kinetic-grid-map-rviz-plugin_1.6.1-0xenial-20190320-181602-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-visualization/ros-kinetic-grid-map-visualization_1.6.1-0xenial-20190320-162423-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map-demos/ros-kinetic-grid-map-demos_1.6.1-0xenial-20190320-183326-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-grid-map/ros-kinetic-grid-map_1.6.1-0xenial-20190320-184931-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

samuelba, 2019-06-14:
I think this is not a problem of grid_map, rather a problem of the ROS package repository.

samuelba, 2019-06-14:
If you got to http://packages.ros.org/ros/ubuntu/pool/main/r/ you will see that all the grid_map packages are there.

closed 2019-06-14 by samuelba