ANYbotics/grid_map #215: The return value of getIndex() is not consistent with the figure.

Houjiafan, 2019-08-15:


Houjiafan, 2019-08-15:
The index is not refreshed if the grid is still in sight, but it supposed to be according to the fugure. Like the left upper corner grid is supposed to be (0,0) anyway.

linuxscofield, 2019-08-28:
In default, the grid_map index not refreshed after map moved. (maybe to save time?). Thus, the right operation should be: **Rearranges data such that the buffer start index is at (0,0)**  after moving the map. For example:
`if (!map.isDefaultStartIndex()) {
    map.convertToDefaultStartIndex();
  }`

maximilianwulf, 2019-10-07:
The data structure is organized as a 2d ringer buffer. This supports the robocentric moving map behavior of the elevation mapping. 
You have to use `map.convertToDefaultStartIndex()` as @linuxscofield already explained.

closed 2019-10-07 by maximilianwulf