ANYbotics/grid_map #216: Conversion from point cloud to grid map elevation issue

brunoeducsantos, 2019-08-29:
Hi,
I developed a conversion from point cloud to grid map elevation layer. Although, for some reason the published point cloud (white) is not totally aligned with height map.
[image]
Any ideas why it could be happening?
Here it is the snippet of code:
// Set up grid map geometry.

  ```
grid_map::Length length = grid_map::Length( max_grid.first- mini_grid.first, max_grid.second- mini_grid.second);
  grid_map::Position position = grid_map::Position((max_grid.first + mini_grid.first) / 2.0,
                                                   (max_grid.second+ mini_grid.second) / 2.0);
  gridMap.setGeometry(length, height_map_resolution_, position);
 
  // Add elevation layer
  gridMap.add(layer);
  gridMap.setFrameId(frame_id_);
  gridMap.setBasicLayers({layer});
  gridMap.move(position);
  // For each voxel, if its elevation is higher than the existing value for the
  // corresponding grid map cell, overwrite it.
  grid_map::Matrix& gridMapData = gridMap[layer];
 
  //Fill data structure with height map
  
  for (int i = 0; i < voxels.size(); i++) {
      grid_map::Position position(voxels[i].x,voxels[i].y);
      grid_map::Index index;
      gridMap.getIndex(position, index);
      grid_map::wrapIndexToRange(index, gridMap.getSize());
      // If no elevation has been set, use current elevation.
      if (!gridMap.isValid(index)) {
        gridMapData(index(0), index(1)) = voxels[i].z;
      }
      // Check existing elevation, keep higher.
      else {
        if (gridMapData(index(0), index(1)) < voxels[i].z) {
          gridMapData(index(0), index(1)) = voxels[i].z;

        }
      }
  }

```
Thanks,
Bruno

maximilianwulf, 2019-10-07:
Hey @brunoeducsantos, where is the origin of the point cloud? Since the elevation map only gets visualized with an (x, y) coordinate it could be that the point cloud has a z offset. 
Were you able to debug it in the meantime?

brunoeducsantos, 2019-10-09:
Hey @maximilianwulf , I was able to debug. It is just the way I assign height to certain points in the grid.
Thanks for your feedback.

closed 2019-10-09 by brunoeducsantos

maximilianwulf, 2019-10-09:
Good to hear, so the values of `voxels[i].z` were wrong or is it actually in the code?

brunoeducsantos, 2019-10-09:
It is in the code. For removing the spikes in the free robot navigable area, I apply a smoothing using knn to remove outliers, setting the height by default to a certain value.

maximilianwulf, 2019-10-10:
Thank you.

Just for completion and future travelers, Grid map also provides smoothing filters, and you could also use median filters from OpenCV to filter out outliers.