ANYbotics/grid_map #217: about SetGridMap.srv

icesit, 2019-09-12:
thank you for the excellent sharing.
when i build the elevation_mapping, it recall a bug that no SetGridMap.h exist.
i install the grid_map by apt install approach, but i am afraid that the apt version is behind the newest version in this github repo, because i can find the SetGridMap.srv in this repo while no SetGridMap.h in my ros install path and i am sure i have install the grid_map_msgs.

Eraledm, 2019-09-30:
I have the same problem when trying to compile elevation_mapping, which version do we have to install in order to get the packages compiled correctly?

maximilianwulf, 2019-10-07:
@hogabrie do you have an idea on that?

LDX596999414, 2019-10-10:
I also have this problem. Who has solved this problem?

maximilianwulf, 2019-10-10:
As @icesit supposed the official release is a bit behind. For now, you have to compile it from source. We will make sure that with the next ROS release grid_map will be updated.
Sorry for the inconvenience.

LDX596999414, 2019-10-10:
> As @icesit supposed the official release is a bit behind. For now, you have to compile it from source. 

> 正如@icesit所认为的那样，正式发布时间有些落后。现在，您必须从源代码进行编译。我们将确保在下一个ROS版本中将更新grid_map。
> 抱歉给你带来不便。

Thank you very much. By the way you said, the problem has been solved.

maximilianwulf, 2019-10-10:
> By the way you said, the problem has been solved.

No, it will be solved.