ANYbotics/grid_map #219: ROS2 Port

jacobhuesman, 2019-12-08:
I'm going to start the process of porting `grid_map` over to ROS2. I noticed some chatter about it on #178, but haven't seen any branches or forks.

@pfankhauser can you create a `ros2` branch that I can submit pull requests against?

maximilianwulf, 2019-12-19:
Hey @jacobhuesman, PRs are very welcome!

Feel free to add PRs to this [ros2 branch](https://github.com/ANYbotics/grid_map/tree/feature/ros2).

SteveMacenski, 2020-05-06:
@maximilianwulf I have some folks working with me on navigation2-land that have interest in working to port over code to ROS2 with the goal of having this replace potentially costmap 2D (or chunks of it).

It might be good to have you join our community slack (navigation2.slack.com) so we can chat about this. I also want to figure out what a reasonable maintenance arrangement looks like if you're not currently using grid_maps in ROS2 (e.g. if PRs come in for 1 version syncing to both, if we need to get things into the ROS2 branch, are there maintainers willing to review and iterate, would it be possible to add me and/or Tully as reviewers strictly for ROS2, etc). 

Alternatively, we could work off of a fork of grid_maps in `ros-perception`, but I'd really like to keep the code with the owners when possible. Deep forks hurts everyone.

maximilianwulf, 2020-05-12:
Hi @SteveMacenski, thanks a lot for the request. We are very interested in helping to port `grid_map` to ROS2. 
Does that mean you already have a plan for the port and we should discuss it on slack?

> are there maintainers willing to review and iterate

Yes, we are willing to help review PRs.

> would it be possible to add me and/or Tully as reviewers strictly for ROS2

Yes, definitely. 

> Alternatively, we could work off of a fork of grid_maps in ros-perception, but I'd really like to keep the code with the owners when possible. Deep forks hurts everyone.

Agreed, we think the same and would like to have it on this repository.

SteveMacenski, 2020-05-12:
We're in the early stages of talking about a port, there's interest shown in helping me redesign the navigation2 environmental representation (e.g. remove costmap 2D and replace it with this + layers built on top). @Marwan99 has expressed interest in helping with the initial port of grid maps for ROS2. From there, its a little fuzzy how we proceed (build a new 3D costmap implementation on top, do some gradient mapping like elevation mapping, do some mixture of both) but this is the first gate to get through. Any thoughts / interest you have on this side are also beneficial. There may be some value in having your environmental models used in navigation2 for your own risk reduction of multiple users playing with it / developing in it. 

We should though discuss more of this in slack if you have thoughts on how that port should happen.  More points if you jump in to help on some of it :wink: . When you join, I'll make a separate channel for this work. 

Please do add @tfoote and I with access. Tully is to make sure that someone can be instated with admin should I get hit with a bus and anybotics stops supporting it. Adding me is so that we can merge PRs for ROS2 only. I have 0 intention of messing with your ROS1 or other stuff. I would though ask that after we get a port in, all PRs should be reflected on both sets so that ROS2 gets the upgrades too until the full shift to ROS2 is made by most companies.

maximilianwulf, 2020-05-13:
Okay, let's discuss it on Slack. 

I will add you both, but I have to check with the team how that is done.

SteveMacenski, 2020-05-13:
Sounds good!

maximilianwulf, 2020-05-14:
Hey @SteveMacenski, can you send me an invite for the slack channel?

Marwan99, 2020-05-14:
https://navigation2.slack.com/

SteveMacenski, 2020-05-14:
Marwan posted the link, if you have issues joining, if you give me your email I can send you a specific invite.

SteveMacenski, 2020-05-20:
@maximilianwulf did you join? We'd like to get started on this this week ideally

maximilianwulf, 2020-05-23:
Hi @SteveMacenski, I seem to have problems logging in. 

Can you send a mail to this email address temporary.maximilian@gmail.com and I will share my slack email address? 

Thanks.

SteveMacenski, 2020-05-26:
Email sent

Marwan99, 2020-06-05:
Ported packages tracker:

- [ ] grid_map
- [ ] grid_map_core
- [ ] grid_map_costmap_2d
- [ ] grid_map_cv
- [ ] grid_map_demos
- [ ] grid_map_filters
- [ ] grid_map_loader
- [ ] grid_map_msgs
- [ ] grid_map_octomap
- [ ] grid_map_pcl
- [ ] grid_map_ros
- [ ] grid_map_rviz_plugin
- [ ] grid_map_sdf
- [ ] grid_map_visualization

SteveMacenski, 2020-06-08:
We should also setup CI for the ROS2 branch. Image pipeline has a good simple example https://github.com/ros-perception/image_pipeline/blob/melodic/.circleci/config.yml

SteveMacenski, 2020-06-24:
I'm going to tdo SDF and Octomap today and tomorrow

SteveMacenski, 2020-07-28:
Huge shout out @Marwan99! Just getting through these like its no one's business!

Marwan99, 2020-09-14:
Hi @wxmerkt!
We are working on porting grid_map to ROS2 and one of the packages have we have a dependency on [octomap_mapping](https://github.com/OctoMap/octomap_mapping). I was just wondering if there is a timeline/plan for porting `octomap_mapping` over to ROS2.
Thanks!

SteveMacenski, 2020-09-14:
@Marwan99 I meant on the octomap mapping ROS2 port ticket :wink:

Marwan99, 2021-01-13:
Is this issue considered completed now?

SteveMacenski, 2021-01-13:
Is there anything we missed? I thought we skipped something (?)

If not, and we're ready to release, then yeah after bloom is run. Is CI passing and everything? Can we do an attempted release? Any dependencies missing?

Marwan99, 2021-01-13:
- One of the demo nodes in grid_map_demos is missing `octomap_server` ([commented](https://github.com/ANYbotics/grid_map/blob/ros2/grid_map_demos/package.xml) out atm) as it is not ported yet.
- `pcl_ros` currently being built from source so I guess that's a non released dependency.
- CI is passing.

maximilianwulf, 2021-01-14:
In my view this issue can be closed. 

@Marwan99 could you open a new one with the remaining points?

SteveMacenski, 2021-01-14:
Agreed, closing!

SteveMacenski, 2021-01-14:
Great job @Marwan99 for single handedly getting this done. That was a big effort and it will be felt for a long time to come.

closed 2021-01-14 by SteveMacenski