ANYbotics/grid_map #220: GridMap layer algorithm

sasupe, 2019-12-09:
Hi @maximilianwulf and team,
Thank You for this great tool.

I am using GridMap to generate a layer similar to elevation, but independent of the pose. Just purely based on the pointcloud at an instant. I am clearing the layer at the end of every time instant.

On implementing this I get the results as following:
[image]

Since the loop just uses the pointcloud (x,y) and finds the corresponding point in the Map and fills it with (z) position, I thought all the points in the lidar cloud should have some elevation value.

So I modified the code, to start the layer with a high negative value at every time instant. Using the  same strategy the result now looks as I had thought (all the pointcloud points have some elevation):
[image]

Since this worked, I added another loop to remove the artificial negative value I had added and replaced it with 'nan', but this lead back to the original results. 
I also tried adding a 'gridmap threshold filter' to remove the artificial negative value but it lead to the original results again.

I believe the underlying buffer algorithm is removing those point everytime. Can you please explain the observations I am seeing, and a part of code you could point me to, so that I can modify this if possible?

Thank You for your time.

AlexReimann, 2019-12-11:
Are you sure it is not just a visualization issue?
Sounds to me like values are simply cut off when visualizing.

maximilianwulf, 2019-12-19:
Hi @sasupe, can you try the pcl loader? https://github.com/ANYbotics/grid_map/tree/master/grid_map_pcl and check if you get the same result?

maximilianwulf, 2019-12-19:
As @AlexReimann suggested, can you check if there are values in the elvation map at that position?

sasupe, 2019-12-25:
Hi! 
Thank You for your inputs. The values are actually present in the elevation map but has an issue while visualizing.

closed 2019-12-25 by sasupe

maximilianwulf, 2020-01-06:
@sasupe thanks for the reply. Could you share the a small rosbag with the grid map for debugging?