ANYbotics/grid_map #222: Adding data points to grid map

Jeromelee95, 2019-12-21:
Hi,

I am trying to input data of position and elevation recorded from my robot into the grid_map package. The output I am trying to achieve will be a 3d map. I have already written the subscriber portion to subscribe to the sensor topics. However, I do not know how to assign the data from the sensor topics to the point on the map. Can anyone advice me how to carry on?

Thanks for your time!

maximilianwulf, 2020-01-06:
Hi @Jeromelee95, isn't [elevation_mapping](https://github.com/ANYbotics/elevation_mapping) doing what you want?

Jeromelee95, 2020-01-06:
Hi @maximilianwulf, thanks for your reply. i dont really understand how elevation_mapping work. Could you guide me through how to work with it?
I have only tried to work with the demo but there are errors running it

maximilianwulf, 2020-01-17:
Hi @Jeromelee95, unfortunately I cannot guide you step by step. But the underlying documentation of elevation map is very descriptive. Have a read [here](https://www.researchgate.net/publication/261949305_Robot-Centric_Elevation_Mapping_with_Uncertainty_Estimates).