ANYbotics/grid_map #223: SubmapIterator does not check map bounds

ahertle, 2019-12-24:
Hi,

I use the SubmapIterator to iterate over the neighbors of a given index: 
```
auto start_index = current_index + grid_map::Index(-1, -1);
for (auto it = grid_map::SubmapIterator(map, start_index, grid_map::Size(3, 3)); !it.isPastEnd(); ++it)
{ ... }
```
When I iterate over the neighbors of a "border" cell, the SubmapIterator returns indices which lie outside the map.

Other iterators (as seen in the demo/readme) respect the grid_map bounds and only return valid indices. I expected the SubmapInterator to behave in the same way. Or am I supposed to check the returned indices myself?

Thanks

closed 2020-01-02 by ahertle

ahertle, 2020-01-02:
Ok, documentation says to check the map boundaries before accessing the index

rahulr-iiitd, 2022-10-23:
@ahertle 
Can you please tell me how you did that? I have a rectangular region that I need to iterate over, and some part of it is outside the map.