ANYbotics/grid_map #227: Build error when compiling from source on ROS for windows.

aditya369007, 2020-03-31:
I am trying to install grid_map on ROS from windows but I am facing the following issue. Can anyone guide me on how to fix this? 

```
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(46): note: see reference to class template instantiation 'boost::arg<1>' being compiled
NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\14.23.28105\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
<== Failed to process package 'grid_map_filters':
  Command '['c:\\opt\\catkin_ws\\devel_isolated\\grid_map_ros\\env.bat', 'nmake']' returned non-zero exit status 2

Reproduce this error by running:
==> cd 'c:\opt\catkin_ws\build_isolated\grid_map_filters' && 'c:\opt\catkin_ws\devel_isolated\grid_map_ros\env.bat' nmake

Command failed, exiting.
```

Thank you!

maximilianwulf, 2020-03-31:
Hi @aditya369007, the package is supported for Ubuntu 18.04 and ROS melodic only at the moment. We can not help. But maybe someone from the community has an idea.

aditya369007, 2020-03-31:
Okay thank you

closed 2020-03-31 by aditya369007