ANYbotics/grid_map #230: Noetic release

jspricke, 2020-06-03:
please provide a noetic release.

jspricke, 2020-06-04:
I've opened a PR with fixes for this in #231.

wxmerkt, 2020-12-09:
Friendly ping - would it be possible to release this package into the ROS buildfarm for Noetic? Thanks!

maximilianwulf, 2020-12-09:
Yes, we are working on it. I guess it will be released soon. Thank you for your patience.

maximilianwulf, 2021-01-06:
Short update on it, we should soon have the Noetic release. Thank you for your patience.

julianraheema, 2021-01-21:
Please we need noetic release.

maximilianwulf, 2021-01-21:
Hey all, all release jobs of grid map have passed. I am currently checking the devel jobs. But I guess the debians are ready to go. 
[image]

Let me know if you have issues and thank you for your patience.

closed 2021-01-21 by maximilianwulf