ANYbotics/grid_map #243: [ROS2] Add back in pcl methods in grid_map_sdf once pcl_ros is available in ROS2

SteveMacenski, 2020-06-25:
currently pcl_ros is not available in ROS2 so the imported headers and conversion functions in grid_map_sdf cannot be fully ported at this time, but all the rest of it is.

Once pcl_ros is ported to ROS2, we can enable this.

maximilianwulf, 2020-06-26:
Is the ROS2 navigation group going to do it? 
Maybe we have to check if the package can work independent of `pcl_ros`.

maximilianwulf, 2020-06-26:
Related to #241 .

SteveMacenski, 2020-06-26:
I'm not sure off hand, one project at a time, pcl_ros isn't a small package. I also am a maintainer on that as well so I can help with getting things pushed through but we simply haven't discussed this in the context of the navigation working group

maximilianwulf, 2020-06-29:
Okay, thanks. I will check if we can have this package independent of `pcl_ros`.

Marwan99, 2021-01-16:
`pcl_ros` is now available in case you decide to continue using it.

I think this also needs to be added to the list in #283.

SteveMacenski, 2021-01-19:
pcl ros for ROS 2 is still not available - we haven't ported it yet

mitsudome-r, 2021-05-13:
@SteveMacenski 
Hi, I'm using this package for my project, and it also helps us if this is released as binary since it will reduce our build time. I was looking to see if there is anything I can help, and I've noticed that there could be alternative solution instead of waiting pcl_ros to be released. Could you take a look at https://github.com/ANYbotics/grid_map/pull/299, and see if the solution works for you?

SteveMacenski, 2021-05-13:
That works for me! Closing ticket since PR resolves

closed 2021-05-13 by SteveMacenski